siliconcompiler/siliconcompiler #788: Using sc-show fails due to pdk file not found

prasadp4009, 2021-12-08:
While testing the flipflop design, I came across issue where if I run the "sc-show -design flipflop", the siliconcompiler throws following error,

--------------------------------------------------------------------------------
| INFO    | job0    | ---          | -   | Setting commandline arguments
| INFO    | job0    | ---          | -   | Displaying /Users/dadu/Projects/chipDesign/build/flipflop/job0/export/0/outputs/flipflop.gds
| INFO    | job0    | ---          | -   | Showing file /Users/dadu/Projects/chipDesign/build/flipflop/job0/export/0/outputs/flipflop.gds
| INFO    | job0    | ---          | -   | Writing manifest to /Users/dadu/Projects/chipDesign/build/_show/sc_manifest.tcl
| INFO    | job0    | ---          | -   | Writing manifest to /Users/dadu/Projects/chipDesign/build/_show/sc_manifest.json
**| ERROR   | job0    | ---          | -   | File ../third_party/pdks/virtual/freepdk45/pdk/r1p0/setup/klayout/freepdk45.lyt was not found
| ERROR   | job0    | ---          | -   | File ../third_party/pdks/virtual/freepdk45/libs/NangateOpenCellLibrary/r1p0/lef/NangateOpenCellLibrary.macro.mod.lef was not found**

It is due to the incorrect path in pdks/*.py files which points to incorrect pdkdir variable.

I found simple workaround to get going where 

1. Clone the siliconcompiler Github Repo. It contains the thirdparty/pdks directory
2. Capture the absolute path of siliconcompiler repo
3. Replace all "../third_party" in all .json files of your project directory with the absolute path of the third_party directory
4. Re-run the sc-show

I believe the third_party dir from pip siliconcompiler package is missing and instructions are missing from the docs. I will work on finding permeant fix for the issue

WRansohoff, 2021-12-08:
Thank you for reporting this, and for proposing a solution!

We're looking into this issue, which happens when the package is installed using `pip`. Like you noted, it's looking for files in the `third_party/` directory which is not currently packaged with the Python wheel.

As an alternative to replacing all of the paths in your JSON manifest files, you should also be able to set an `SCPATH` environment variable pointing to the siliconcompiler repository's root directory. [You can read a bit more about how the system looks for files here](https://docs.siliconcompiler.com/en/latest/user_guide/targets.html); the logic used by the `target()` method is also used wherever we try to find a file using a path from the schema.

We've also recently [pushed a change](https://github.com/siliconcompiler/siliconcompiler/commit/d1543f41777171892b63b20beef94356cde74e06) to make these PDK files optional, and we're hoping to release those fixes soon.

nmoroze, 2021-12-08:
We've released a fix to this issue in v0.1.1, which is now uploaded to PyPI. You can update your installation with: `pip install --upgrade siliconcompiler`. 

There will still be a couple error messages printed, but KLayout should successfully display the resulting GDS now. We'll include a further fix to silence those errors in a future release! Closing for now, but please feel free to re-open the issue if the fix doesn't work for you.

closed 2021-12-08 by nmoroze