siliconcompiler/siliconcompiler #792: server details

ratul619, 2021-12-09:
Hi , 

May be a very naïve question  : 

In the venv setup : 

 sc-configure
Remote server address: your-server
Remote username: your-username
Remote password: your-key
Remote configuration saved to: /home/<USER>/.sc/credentials

How to get the server address ? ipconfig ? Which field in ipconfig >? 
Also name and password would be new ? 

Thanks

WRansohoff, 2021-12-09:
Good questions - at the moment, our API server is located at https://server.siliconcompiler.com. To get a username and password for that server, you can [sign up for our cloud beta here](https://www.siliconcompiler.com/beta).

Sorry for the confusion; it sounds like we should consider updating our documentation pages to point to that signup page.

ratul619, 2021-12-09:
@WRansohoff  Thanks. But seems the server not open for general resigistration

WRansohoff, 2021-12-14:
We are restricting sign-ups to a handful of email domains and TLDs (such as *.edu) for this very early beta period - sorry for the inconvenience. But if you don't have an email address which the beta sign-up form will send credentials to, you can also install the open-source tools locally and build designs on your machine.

For now, we have setup scripts to automate the installation process on recent Debian-based platforms; you can find them in the [`setup/`](https://github.com/siliconcompiler/siliconcompiler/tree/main/setup) directory. I believe that the default ASIC flow requires `surelog`, `yosys`, `openroad, and `klayout`, but I could be forgetting a tool. Unfortunately, we do not currently support local builds on Windows.

closed 2021-12-23 by aolofsson