siliconcompiler/siliconcompiler #821: setup: multi-job make

lukasc-ch, 2022-01-02:
It would be great to replace the hard-coded job count (typically 1 or 2) of the make calls in the setup scripts with `ncore` to allow >10x faster build times on many systems. 

If this is desirable, I can create a PR. Similarly, while at it, I could also create a matching Dockerfile for automatic environment setup.

nmoroze, 2022-01-03:
Thanks for the suggestion! I think a PR to automatically parallelize the setup scripts would be nice. We'll hold off on the Dockerfile for now, we don't want to add this until we can commit to testing and maintaining it.

aolofsson, 2022-01-03:
The hard coded count you saw might be fore tools that aren't fully implemented. Most heavy duty tools in SC already have dynamic checking of core count through a python interface. What would `ncore` add that the current approach shown below doesn't have?

# openroad setup
```python
chip.set('eda', tool, 'threads', step, index, os.cpu_count(), clobber=clobber)
```

We are definitely open to ideas around docker and automatic setup! Right now we have just captured the ubuntu setup dependancies as place holders for our own benefit. https://github.com/siliconcompiler/siliconcompiler/tree/main/setup

Before submitting a PR, can you describe what you had in mind? How does it relate/compare to the hdl containter effort?

https://hdl.github.io/containers/

aolofsson, 2022-03-30:
Closing for lack of activity.

closed 2022-03-30 by aolofsson