siliconcompiler/siliconcompiler #831: Heartbeat seems to hang on OpenROAD insert tapcells step

suzizecat, 2022-01-05:
Hello, first time user here !
Using Python 3.8 on Ubuntu 20.04,

Following the instructions on https://docs.siliconcompiler.com/en/latest/user_guide/installation.html#siliconcompiler for installation in a fresh virtual environment and https://docs.siliconcompiler.com/en/latest/user_guide/quickstart.html for a quick start, the run seems to hang on what I believe is the floorplanning job whith the following console output: 

```text
...
| INFO    | job0    | floorplan    | 0   | openroad -no_init -exit /media/Data/Linux/eda/siliconcompiler/siliconcompiler/tools/openroad/sc_apr.tcl
OpenROAD 1.1.0 GITDIR-NOT
License GPLv3: GNU GPL version 3 <http://gnu.org/licenses/gpl.html>

This is free software, and you are free to change and redistribute it
under certain conditions; type `show_copying' for details. 
This program comes with ABSOLUTELY NO WARRANTY; for details type `show_warranty'.
Notice 0: Reading LEF file:  /media/Data/Linux/eda/siliconcompiler/third_party/pdks/virtual/freepdk45/pdk/r1p0/apr/freepdk45.tech.lef
Notice 0:     Created 22 technology layers
Notice 0:     Created 27 technology vias
Notice 0: Finished LEF file:  /media/Data/Linux/eda/siliconcompiler/third_party/pdks/virtual/freepdk45/pdk/r1p0/apr/freepdk45.tech.lef
Notice 0: Reading LEF file:  /media/Data/Linux/eda/siliconcompiler/third_party/pdks/virtual/freepdk45/libs/NangateOpenCellLibrary/r1p0/lef/NangateOpenCellLibrary.macro.mod.lef
Notice 0:     Created 134 library cells
Notice 0: Finished LEF file:  /media/Data/Linux/eda/siliconcompiler/third_party/pdks/virtual/freepdk45/libs/NangateOpenCellLibrary/r1p0/lef/NangateOpenCellLibrary.macro.mod.lef
Error: sc_apr.tcl, 172 set_wire_rc -clock is not a known keyword or flag.
Error: sc_apr.tcl, 173 set_wire_rc -signal is not a known keyword or flag.
Error: sc_apr.tcl, 177 invalid command name "set_placement_padding -global -left 2 -right 2"
(0, 0) (30.4, 30.4)
(1.9, 1.9) (28.499999999999996, 28.499999999999996)
 0  0   30.4  30.4 
 1.9  1.9   28.499999999999996  28.499999999999996 
Info: Added 19 rows of 140 sites.
Error: sc_floorplan.tcl, 101 invalid command name "make_tracks metal1 -x_offset 0.095 -x_pitch 0.19 -y_offset 0.07 -y_pitch 0.14"
Running tapcell...
Step 1: Cut rows...
---- Macro blocks found: 0
---- #Original rows: 19
---- #Cut rows: 0
Step 2: Insert endcaps...
---- #Endcaps inserted: 38
Step 3: Insert tapcells...
```

You can note that there are some errors due to unknown directives which seems concerning to me, but I wouldn't know.

Would you have some clues on what might be wrong ? (This also could be a setup issue, but I can't put my finger on it).
For reference, find attached my full testcase, just run `python3 heartbeat.py` [heartbeat.zip](https://github.com/siliconcompiler/siliconcompiler/files/7814491/heartbeat.zip)
 
I tried using the SKY130 target too, without difference.

Best regards,
Julien FAUCHER

nmoroze, 2022-01-05:
Thanks for trying out SC and for the detailed bug report!

I think the issue is likely related to an incompatible version of OpenROAD. We recommend you install OpenROAD commit `d1c7001ad`. This is the version bundled in our `third_party/tools/openroad` submodule.

suzizecat, 2022-01-06:
Thanks for the feedback ! 

I previously tried to fiddle with OpenROAD/Lane and had a remnant openroad install laying around. Therefore, I had no message to let me know that openroad wasn't properly installed. You might want to add a note about that somewhere.

It worked just fine with a proper install, thanks again :slightly_smiling_face:

closed 2022-01-06 by suzizecat

nmoroze, 2022-01-06:
No worries, glad to hear that worked!