siliconcompiler/siliconcompiler #835: Summary to file

qarlosalberto, 2022-01-10:
Is there a way to get the summary as string, similar to:

```
chip.run()                                    
summary = chip.get_summary()     
```
So it could be saved in a file.

nmoroze, 2022-01-10:
I don't believe we have a quick and easy way to do this yet. One option as a workaround could to copy-paste [the summary implementation from core.py](https://github.com/siliconcompiler/siliconcompiler/blob/main/siliconcompiler/core.py#L2463-L2585) into your own code, and replace the `print()` calls at the end with a single `return df.to_string()`. Then you could call this directly from your code. 

This is a good idea though, and we can take it under consideration as a feature request!

aolofsson, 2022-01-11:
Depends on what you want to do with the summary? All the data is in the .json file, which is dumped with the write_manifest() method. The summary function is just looping through the metrics dictionary to print put the values to screen. The same thing could be done eaily in Javascript or any other language that can parse json.

qarlosalberto, 2022-01-11:
Got it! I didn't know the write_manifest() function. It works for me. Thanks both!

aolofsson, 2022-01-11:
Fantastic!

closed 2022-01-11 by aolofsson