siliconcompiler/siliconcompiler #836: Error to set build directory

qarlosalberto, 2022-01-10:
I have an error when I set the output directory (I'm using https://server.siliconcompiler.com as remote server). Is it correct?:

```
home_dir = os.path.expanduser("~")
work_root = os.path.join(home_dir, '.teroshdl', 'build')
chip.set('dir',work_root)
```

```
| INFO    | job0    | ---          | -   | Running in /home/carlos/.teroshdl/build/heartbeat/job0/import/0
| INFO    | job0    | ---          | -   | /usr/local/lib/python3.8/dist-packages/siliconcompiler/tools/surelog/bin/surelog -parse /home/carlos/.teroshdl/build/heartbeat/job0/import/0/outputs/heartbeat_2c561ab2d18f3e79eb2843381c8a32cf.v -top heartbeat +libext+.sv
| INFO    | job0    | ---          | -   | Writing manifest to /home/carlos/.teroshdl/build/heartbeat/job0/import/0/outputs/heartbeat.pkg.json
| ERROR   | job0    | ---          | -   | Error: value for ['dir'] must contain only alphanumeric characters and underscores.
| ERROR   | job0    | ---          | -   | Error starting remote job run; quitting.
```

nmoroze, 2022-01-10:
For security reasons we don't currently let you set a build directory name that contains characters other than alphanumerics or underscores when running jobs on the remote. It looks like you're setting an absolute directory as a build dir, which is going to include slashes. I'd recommend setting a single name for your build directory (or just using the default of `build/`) and then adding additional code to copy the build results from here into a different directory on your filesystem if needed. Does that make sense?

qarlosalberto, 2022-01-11:
I was confused, I thought it was the full path to the output directory. Thanks!

closed 2022-01-11 by qarlosalberto