siliconcompiler/siliconcompiler #837: Dockerfile or VM-based image for cloud machines

sbates130272, 2022-01-10:
Great tool. Any chance that as well as adding the install scripts for all the tools we could add a Dockerfile or Vagrant (or similar) file for a machine container/image with all the tools pre-installed?

aolofsson, 2022-01-11:
Thanks! With over 20 executable supported, bundling consistent VMs is a big challenge. There are some efforts that we are watching like https://hdl.github.io/containers/.

In the short term, we have captured the setup scripts here.
https://github.com/siliconcompiler/siliconcompiler/tree/main/setup

For now, the easiest path is to run with the -remote switch (zero install).

sbates130272, 2022-01-12:
Thanks @aolofsson. I recommend you take a look at Packer which enables a very consistent VM or simply use a Dockerfile creation process.

aolofsson, 2022-01-12:
Thanks for the tip, will do!

closed 2022-01-12 by aolofsson