siliconcompiler/siliconcompiler #849: Yosys doesn't use UHDM and therefore doesn't completely accept SV output from Surelog

suzizecat, 2022-01-17:
Hello,

The `syn` process uses `read_verilog -sv` instead of `read_uhdm` (available as "unofficial" command from https://github.com/antmicro/yosys with the uhdm-plugin branch.)

This implies that specific SV structures such as SV interface are not accepted by SiliconCompiler.
Moreover, this should degrade performances as the aggregated verilog file used instead of UHDM database have to be re-parsed again.

suzizecat, 2022-01-17:
This can be fixed by adding 
```python
    shutil.copy("slpp_all/surelog.uhdm",f"outputs/{design}.uhdm")
```
To `surelog.py` and to add the following function to the chip definition file : 

```python
def update_surelog_options(chip : siliconcompiler.Chip):
    step = "import"
    index = "0"
    tool = "surelog"
    chip.add('eda', tool, 'output', step, index, chip.get('design') + '.uhdm')
    chip.add('eda', tool, 'option', step, index, '-elabuhdm')

def update_yosys_options(chip : siliconcompiler.Chip) :
    step = "syn"
    index = "0"
    tool = "yosys"
    chip.add('eda', tool, 'option', step, index, '-m uhdm')
 ```
 
 since, in my situation, I have `uhdm` support as a yosys module.

nmoroze, 2022-01-18:
Thanks for pointing this out. However, this was an intentional design decision on our part. We experimented a bit with the Yosys read_uhdm frontend and decided it wasn't quite ready for our purposes. Surelog's place in our build flow is solely to gather all input sources into a single pickled (System)Verilog file, which we then pass along directly to Yosys. In order to support SystemVerilog constructs that Yosys does not natively support, we have an 'sv' frontend option that enables an additional conversion pass using sv2v. You can enable it like so: `chip.set('frontend', 'sv')`.

closed 2022-01-18 by nmoroze

suzizecat, 2022-01-18:
Thanks for your feedback.
I understand clearly your motivation as I regularly find issues with Surelog/UHDM :sweat_smile: 

Just to know, is the sv2v step supposed to appears in the console output at some point ?
I don't see any real change (I might have missed it though) in the output except for the aggregated `design.sv` file being a `.sv` instead of a `.v` . The command in yosys is `read_verilog -sv` in both cases. (And seems to fails in both situations)

suzizecat, 2022-01-18:
Nevermind, I didn't knew about the fact that you shall use those command before calling `chip.target` . It still doesn't work, but at least seems to be trying !