siliconcompiler/siliconcompiler #850: Misuse of `sv2v --write` parameter

suzizecat, 2022-01-18:
Hi !
When using `systemverilog` as a frontend, siliconcompiler issues the command 
```
sv2v inputs/core_top.v --write=outputs/core_top.v
```
Whereas the `write` argument is should be used to provide an output mode rather than an output file
```text 
sv2v --help
[...]
 -w --write=MODE           How to write output; default is 'stdout'; use
                           'adjacent' to create a .v file next to each input
```

nmoroze, 2022-01-18:
I think you might need a newer version of sv2v. The --write flag was changed at some point to allow it to be used to specify an output file (from the README.md on current top-of-tree):

```
  -w --write=MODE/FILE      How to write output; default is 'stdout'; use
                            'adjacent' to create a .v file next to each input;
                            use a path ending in .v to write to a file
```

I believe the functionality was added in this commit: https://github.com/zachjs/sv2v/commit/404385b00f45fa553184757718028e213448768d.

closed 2022-01-18 by nmoroze

suzizecat, 2022-01-18:
Indeed, thanks !
It seems that sv2v packaged with various distribution of surelog is not up to date... 
The whole flow (seems to) works now !