siliconcompiler/siliconcompiler #852: what does the ERROR STA-0406 mean?

rrossi94, 2022-01-21:
Hi to all!

First to all congrats to @aolofsson and team for this great work!

Having said that, I am having an error with one of the .tcl file, while executing the heartbeat example (See https://docs.siliconcompiler.com/en/latest/user_guide/quickstart.html). I created the python script that says the quickstart page but when I try to run it (by `python3 heartbeat.py` ), I get the following error: 

```
OpenROAD v2.0-2686-gb6b65120e 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: /home/rrossi/siliconcompiler/third_party/pdks/virtual/freepdk45/pdk/r1p0/apr/freepdk45.tech.lef
[INFO ODB-0223]     Created 22 technology layers
[INFO ODB-0224]     Created 27 technology vias
[INFO ODB-0226] Finished LEF file:  /home/rrossi/siliconcompiler/third_party/pdks/virtual/freepdk45/pdk/r1p0/apr/freepdk45.tech.lef
[INFO ODB-0222] Reading LEF file: /home/rrossi/siliconcompiler/third_party/pdks/virtual/freepdk45/libs/NangateOpenCellLibrary/r1p0/lef/NangateOpenCellLibrary.macro.mod.lef
[INFO ODB-0225]     Created 134 library cells
[INFO ODB-0226] Finished LEF file:  /home/rrossi/siliconcompiler/third_party/pdks/virtual/freepdk45/libs/NangateOpenCellLibrary/r1p0/lef/NangateOpenCellLibrary.macro.mod.lef
[INFO ODB-0127] Reading DEF file: inputs/heartbeat.def
[INFO ODB-0128] Design: heartbeat
[INFO ODB-0130]     Created 3 pins.
[INFO ODB-0131]     Created 65 components and 235 component-terminals.
[INFO ODB-0133]     Created 40 nets and 105 connections.
[INFO ODB-0134] Finished DEF file: inputs/heartbeat.def
[INFO DPL-0001] Placed 174 filler instances.
[ERROR STA-0406] global_route positional arguments not supported.
Error: sc_route.tcl, 37 STA-0406
| WARNING | job0    | route        | 0   | Command failed. See log file /home/rrossi/siliconcompiler/heartbeat_example/build/heartbeat/job0/route/0/route.log
| ERROR   | job0    | route        | 0   | Halting step 'route' index '0' due to errors.
| ERROR   | job0    | dfm          | 0   | Halting step due to previous error in route0
| ERROR   | job0    | dfm          | 0   | Halting step 'dfm' index '0' due to errors.
| ERROR   | job0    | export       | 0   | Halting step due to previous error in dfm0
| ERROR   | job0    | export       | 0   | Halting step 'export' index '0' due to errors.
| ERROR   | job0    | ---          | -   | Run() failed, exiting! See previous errors.
```

Someone knows what the error `[ERROR STA-0406] global_route positional arguments not supported.` stands for?

Thanks in advance,

Ramiro Rossi.

nmoroze, 2022-01-23:
Hi Ramiro,

I think this is a problem with your OpenROAD version -- it seems to be newer than our officially supported version, and it looks like there was a backwards-incompatible API change. We recommend using OpenROAD commit d1c7001ad.

closed 2022-01-23 by nmoroze

cansudemirkiran, 2022-05-04:
Hi, 
I use the conda setup and the provided [environment.yml](https://github.com/siliconcompiler/siliconcompiler/blob/main/setup/conda/environment.yml) file and I get the same error as above. I was wondering if it's possible to fix this by specifying the openroad version in the environment.yml file. Thanks.

mithro, 2022-05-04:
The docs @ https://docs.conda.io/projects/conda/en/latest/user-guide/tasks/manage-environments.html#creating-an-environment-file-manually specify how to set the OpenROAD version to a specific version. It already looks like a specific version of Surelog is being required.

You can then use a tool [`conda-lock`](https://github.com/conda-incubator/conda-lock) to automatically update the versions and send pull requests with the updated values.

cansudemirkiran, 2022-05-05:
Thanks for the explanation! What is the required version in this case for openroad? For example, for surelog it shows surelog==0.0_3379_g484687d13. I believe the last part is the commit number? What whould it be for openroad?