siliconcompiler/siliconcompiler #865: Netgen LVS run failed for example design because of mismatch in number of nets

jarvis-lorry, 2022-02-09:
I've run the example design, heartbeat, with DRC/LVS enabled and get a mismatch in the number of nets, meaning that LVS fails.  I see a similar problem with other designs I've run.

Circuit 1 contains 34 devices, Circuit 2 contains 34 devices.
Circuit 1 contains 53 nets,    Circuit 2 contains 104 nets. *** MISMATCH ***

Result: Netlists do not match.
Logging to file "outputs/heartbeat.lvs.out" disabled
LVS Done.

I've attached the lvs.log.
[lvs.log](https://github.com/siliconcompiler/siliconcompiler/files/8036593/lvs.log)
.

nmoroze, 2022-02-15:
Could you please attach the outputs/heartbeat.lvs.out file? I think that will be helpful for debugging. It would also be helpful to see the Python code or CLI invocation you're using to drive SC.

I was able to get LVS clean on heartbeat using Netgen 1.5.192. I've noticed issues with different errors cropping up depending on version, so could also be worth trying to downgrade to that version and see if that helps the problem.

jarvis-lorry, 2022-02-15:
The LVS mismatches appear to come from the lack of power pins in the Verilog netlist file.
[heartbeat.lvs.out.txt](https://github.com/siliconcompiler/siliconcompiler/files/8074362/heartbeat.lvs.out.txt)

jarvis-lorry, 2022-02-15:
I'm running with Netgen 1.5.219 and I'm running with the heartbeat example.  Attached is my .py file.
[heartbeat.py.txt](https://github.com/siliconcompiler/siliconcompiler/files/8074422/heartbeat.py.txt)

jarvis-lorry, 2022-02-15:
One other question: when will DRC/LVS checks be available for PDKs other than sky130A?  Is there a reason that Magic/Netgen cannot be used with other types of PDKs?

nmoroze, 2022-02-15:
Got it, thanks for sending this over! As a workaround I'd recommend trying Netgen 1.5.192 for now, I think this issue is related to #480. We still have to investigate why this behavior is different between Netgen versions. 

To answer your other question, we don't yet have Magic runsets (the file(s) that codify the DRC rules) for the other PDKs. There may be some available somewhere (or someone could create them), but we don't currently have that on the roadmap. We'd certainly appreciate this as a contribution though!

jarvis-lorry, 2022-02-16:
I realized that my PATH was incorrect and was pointing to an old version of Netgen.  I've installed Netgen 1.5.192 and now I can get LVS to pass. Good news, though I noticed that the SiliconCompiler run didn't end cleanly.  There's an type error reported.

---------------------------------------------------------------------------------------------------------------------------------------
| INFO    | job0    | ---          | -   | Showing file None
Traceback (most recent call last):
  File "heartbeat.py", line 16, in <module>
    main()
  File "heartbeat.py", line 13, in main
    chip.show()                                   # show layout file
  File "/home/lekan/.local/lib/python3.8/site-packages/siliconcompiler/core.py", line 3582, in show
    filepath = os.path.abspath(filename)
  File "/usr/lib/python3.8/posixpath.py", line 374, in abspath
    path = os.fspath(path)
TypeError: expected str, bytes or os.PathLike object, not NoneType

nmoroze, 2022-03-08:
Thanks for pointing that out - #907 should fix once merged.

PR #907 (2022-03-08): Some show() bugfixes
- For now, hardcode last step as export. Otherwise, this breaks when running Skywater verification steps.
- Make extra_options use 'None' sentinel instead of list, using list in kwarg can cause troubles due to Python quirk.
- Make _makecmd accept extra_options so we don't have to modify schema. This results in trouble if we want to make multiple calls to show().
- Clean up logging a bit so we don't print "None"

Closes #865.
closed 2022-03-08 by aolofsson