siliconcompiler/siliconcompiler #876: GSOC 2022 Project Ideas

nmoroze, 2022-02-21:
SiliconCompiler is an open-source Python-based build system for ASIC design tools.

## Several pipecleaner designs

For this project, we would like to build a variety of small but realistic digital designs using the SiliconCompiler Python API, and document those build scripts in a similar manner to [our “ZeroSoc” tutorial](https://docs.siliconcompiler.com/en/latest/tutorials/zerosoc.html).

Difficulty: Intermediate

## Multi-target LiteX design

For this project, we would like to generate a minimal SoC using a platform such as LiteX, and use SiliconCompiler to build the design for 3 targets: local simulation, FPGA simulation, and Sky130 ASIC layout.

Difficulty: Intermediate

## Mixed-language RTL support

In order to ease the process of integrating IP from different sources, we would like to support multiple languages in the designs which SiliconCompiler processes, using open-source tools. (e.g., SystemVerilog and VHDL source files in the same project)

Difficulty: Advanced

## ZeroSoC Simulation using Verilator

Being able to run simulations using the same fileset used for tapeout is crucial for ensuring simulation is representative of a final product. We would like to extend SiliconCompiler’s Verilator driver so that SC can be used to drive simulation as well as compilation. The goal of this project would be to drive a Verilator simulation of our ZeroSoC SoC using SC, running it on a fixed input trace and validating its outputs against a golden reference.

Difficulty: Advanced

## RISC-V Benchmark Suite

The RISC-V community has developed a number of high quality CPU projects, but there is limited open data on the power, speed, and area for these projects when implemented in ASIC processes using open source tool chains. This project will take a number of popular open source RISC-V cores, run them through Skywater130 and ASAP7 technologies using the SiliconCompiler asicflow (Surelog, Yosys, Openroad, Klayout) and extract metrics and layout views for comparison. 

Difficulty: Intermediate

## Chisel or Bluespec RISC-V Core

SiliconCompiler includes basic frontends for supporting alternate HDLs such as Chisel and Bluespec, but they have only been tested on simple single-file designs. We would like to improve these frontends and demonstrate that they can be used to support realistic designs. For this project, we would use SiliconCompiler to compile a small Bluespec or Chisel RISC-V core for an open PDK. 

Difficulty: Intermediate

mithro, 2022-02-21:
Only a limited number of organizations get into Google Summer of Code. You might consider having your projects under the FOSSi Foundation -> https://github.com/fossi-foundation/fossi-foundation.github.io/pull/52

aolofsson, 2022-02-21:
Good point! I will check to see if fossie is interested in sponsoring these ideas.