siliconcompiler/siliconcompiler #897: Surelog paths need to be stripped.

hohe, 2022-03-07:
On a side note, modern (built today) Surelog seems to have broken the `post_process` step by adding double quotes in the paths written in `file.lst` so you might want to add `path = path.strip('"')` at the beginning of the `for path in filelist.read().split('\n'):` loop. (around line 140 in `surelog.py` )

_Originally posted by @suzizecat in https://github.com/siliconcompiler/siliconcompiler/issues/849#issuecomment-1015402219_

This note was really helpful. I'm also using a fresh compiled surelog, I had the same error and after adding
path = path.strip('"')
it vanished.

mithro, 2022-03-07:
If this is a bug in Surelog, please report a bug to Surelog and we'll endeavour to fix them....

hohe, 2022-03-07:
I'm not sure if this is a bug in Surelog. What changed is that their slpp_all/file.lst which lists the *.v files now prints double quotes around the paths.

nmoroze, 2022-03-08:
Closing; fixed in #898 (thanks for the contribution)! We aren't updating the "official" bundled Surelog yet, but this change should be harmless on old versions.

closed 2022-03-08 by nmoroze