siliconcompiler/siliconcompiler #901: Missing NangateOpenCellLibrary_typical.lib

suppamax, 2022-03-08:
Hello everyone,

I've installed (successfully, I think) SiliconCompiler via `pip`
```
#  python3 -m pip show siliconcompiler
Name: siliconcompiler
Version: 0.7.0
Summary: A compiler framework that automates translation from source code to silicon.
Home-page: https://siliconcompiler.com
Author: Andreas Olofsson
Author-email: andreas.d.olofsson@gmail.com
License: Apache License 2.0
Location: /usr/local/lib/python3.8/dist-packages
Requires: aiohttp, cryptography, defusedxml, distro, graphviz, Jinja2, netifaces, numpy, pandas, PyYAML, requests
Required-by: 
```
but when I try to process my design
```
sc top.v -constraint constraints/constraints.sdc -target "freepdk45_demo"
```
I get this error message
```
[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 26
[   NOTE] : 31
| INFO    | job0    | import       | 0   | [  ERROR] : 0
| INFO    | job0    | import       | 0   | [WARNING] : 26
| ERROR   | job0    | syn          | 0   | File ../third_party/pdks/virtual/freepdk45/libs/nangate45/r1p0/lib/NangateOpenCellLibrary_typical.lib was not found
| ERROR   | job0    | syn          | 0   | Required file keypath ['library', 'nangate45', 'nldm', 'typical', 'lib'] can't be resolved.
| ERROR   | job0    | syn          | 0   | File ../third_party/pdks/virtual/freepdk45/libs/nangate45/r1p0/techmap/yosys/cells_latch.v was not found
| ERROR   | job0    | syn          | 0   | Required file keypath ['library', 'nangate45', 'techmap', 'yosys'] can't be resolved.
| ERROR   | job0    | syn          | 0   | Fatal error in check_manifest()! See previous errors.
```
I did try to find `NangateOpenCellLibrary_typical.lib` in my system, but it was nowhere.

Any suggestion?

suppamax, 2022-03-08:
looks like some tools had not been installed.

closed 2022-03-08 by suppamax

nmoroze, 2022-03-08:
Also note that for this error in particular, you'll need clone the repo to get access to the third party PDKs (which aren't bundled with the wheel).

```bash
git clone https://github.com/siliconcompiler/siliconcompiler
export SCPATH=$PWD/siliconcompiler/siliconcompiler # this line points SC to the PDK/library data
```

If you're interested, signing up for our cloud beta can alleviate the need for the tool and PDK setup steps: https://www.siliconcompiler.com/beta.