siliconcompiler/siliconcompiler #904: KLayout published on PyPi

mithro, 2022-03-08:
I saw another project talk about this and thought it could be useful for Silicon Compiler in some way?
 * https://www.klayout.org/klayout-pypi/
 * https://pypi.org/project/klayout/

aolofsson, 2022-03-08:
Thanks, that would be very useful! We have surelog packaged in our build, but users have to install klayout manually. We'll take a look! cc @nmoroze @WRansohoff

WRansohoff, 2022-03-08:
Thanks for the tip!

We might be able to use this for performing the GDS-streaming step in a Windows or tool-less environment. I don't think that [def2stream.py](https://github.com/siliconcompiler/siliconcompiler/blob/main/siliconcompiler/tools/klayout/def2stream.py) requires any of KLayout's UI-related API calls.

It looks like we would probably still need a full KLayout installation in cases where a GUI or headless window is required, like `sc-show` or the screenshot-generation logic. But if we end up installing the EDA tools using a conda environment as suggested in #906, this could also help make up for the absence of KLayout in the [conda-eda](https://github.com/hdl/conda-eda) repository.

mithro, 2022-03-10:
BTW We have been looking at what we would need to do to publish Surelog/UHDM via PyPi. It would be great to have your assistance with making that happen if you are already doing that inside silicon compiler!

mithro, 2022-03-10:
Have you also seen http://yowasp.org/ ? There are `yowasp-yosys` and `yowasp-nextpnr-xxxx` packages uploaded to PyPi. They run a bit slower than natively compiled packages but do work across Windows / Linux / Mac and all architectures if I understand correctly?

nmoroze, 2022-03-10:
> BTW We have been looking at what we would need to do to publish Surelog/UHDM via PyPi. It would be great to have your assistance with making that happen if you are already doing that inside silicon compiler!

We'd be very interested in helping to make this happen! Managing the Surelog bundling is a lot of added complexity in our build/release process, so it'd be nice to offload that to the main project (and also let others benefit from it outside of SC). We're also interested in being able to inspect UHDM from within Python, so that would be helpful as well.

> Have you also seen http://yowasp.org/ ? There are yowasp-yosys and yowasp-nextpnr-xxxx packages uploaded to PyPi. They run a bit slower than natively compiled packages but do work across Windows / Linux / Mac and all architectures if I understand correctly?

I haven't personally, but this is also a great pointer! Thanks, we'll take a look.

closed 2024-05-23 by gadfort