siliconcompiler/siliconcompiler #905: How would something like GDSFactory be integrated into Silicon Compiler?

mithro, 2022-03-08:
GDSFactory (https://github.com/gdsfactory/gdsfactory and https://gdsfactory.github.io/gdsfactory/) is a library for doing the creation of silicon structures that seems to be used heavily in the photonics world. It would be awesome if blocks created by GDSFactory could be generated and integrated into a Silicon Compiler workflow.

BTW GDSFactory also has a pretty nice GDS viewer used in their Python notebook tutorials which could be great to steal for your own documentation, see https://gdsfactory.github.io/gdsfactory/notebooks/06_component_circuit_mask.html for example.

aolofsson, 2022-03-08:
Wow, great project! Thanks for sharing, lots of potential for sharing infrastruture, especially with the floorplanning API from @nmoroze 

https://docs.siliconcompiler.com/en/latest/tutorials/zerosoc.html

aolofsson, 2022-04-14:
Tracking and will use if it makes sense. Added to the awesome tools list.

closed 2022-04-14 by aolofsson