siliconcompiler/siliconcompiler #930: OpenROAD sc_display.tcl in consistent with schema

arlpetergadfort, 2022-03-11:
https://github.com/siliconcompiler/siliconcompiler/blob/a202264ed29cbb960d85fad3be29cd071d4e66b2/siliconcompiler/tools/openroad/sc_display.tcl#L37

Should have stackup as well.

Additionally, it would be nice if the liberty files were loaded along with the sdc file.

nmoroze, 2022-03-11:
I think we might actually be able to just get rid of sc_display.tcl and reuse sc_apr.tcl setup here. I'll do some experimenting.

nmoroze, 2022-03-11:
Opened a PR implementing this. One thing to keep in mind is that show spawns a new step, meaning any SDC files fed in on a per-step basis using the 'read', 'sdc' schema won't be loaded. However, things fed in via 'constraints' will be. I'm not sure if this works for your use case, I also think this subtle distinction might trip people up.

nmoroze, 2022-03-14:
#933 is merged, closing.

closed 2022-03-14 by nmoroze