siliconcompiler/siliconcompiler #931: Klayout formatting failure

arlpetergadfort, 2022-03-11:
https://github.com/siliconcompiler/siliconcompiler/blob/a202264ed29cbb960d85fad3be29cd071d4e66b2/siliconcompiler/tools/klayout/def2stream.py#L225
https://github.com/siliconcompiler/siliconcompiler/blob/a202264ed29cbb960d85fad3be29cd071d4e66b2/siliconcompiler/tools/klayout/def2stream.py#L226

These two lines keep messing me up with an invalid syntax error. changing them to `'{0}.def'.format(design)`  appears to work, not sure why the current one doesn't work.

arlpetergadfort, 2022-03-11:
A similar issue comes up: https://github.com/siliconcompiler/siliconcompiler/blob/a202264ed29cbb960d85fad3be29cd071d4e66b2/siliconcompiler/tools/klayout/klayout_show.py#L94

arlpetergadfort, 2022-03-11:
klayout version 0.27.2

nmoroze, 2022-03-11:
That style of format string is new in Python 3.6, so my best guess is that there's either an older version of Python bundled with your KLayout install or it's finding an older version of Python from somewhere on your system. What platform are you using and how did you download KLayout?

arlpetergadfort, 2022-03-11:
> That style of format string is new in Python 3.6, so my best guess is that there's either an older version of Python bundled with your KLayout install or it's finding an older version of Python from somewhere on your system. What platform are you using and how did you download KLayout?

@nmoroze that formatting style works everywhere else in Silicon Compiler, so it probably is something bundled in klayout. Platform is RHEL7, klayout from klayout from: https://www.klayout.org/downloads/CentOS_7/klayout-0.27.2-0.x86_64.rpm
The machine has python3.6.8 installed (no other versions installed)

arlpetergadfort, 2022-03-11:
@nmoroze looking at the change log for layout it looks looks CentOS7 started supporting python3 in 27.5, I'll try to update and see if that solves it, if so you may want to limit the version to atleast 0.27.5

arlpetergadfort, 2022-03-11:
@nmoroze moving up to 0.27.8 appears to have worked.

nmoroze, 2022-03-14:
Sounds good! We currently have "supported" version set to 0.27.8 (https://github.com/siliconcompiler/siliconcompiler/blob/main/siliconcompiler/tools/klayout/klayout.py#L93), but we don't yet have version checking enabled since it's a bit too strict. It's on our near-term roadmap to add '>=' version checking, at which point we can re-enable version checks.

Closing for now.

closed 2022-03-14 by nmoroze