siliconcompiler/siliconcompiler #969: Clearing schema entries after run()

nmoroze, 2022-03-30:
An issue came up while I was working on #965 that I think points at a larger conceptual issue. The problem comes when we perform multiple `run()` calls in succession on the same chip object.

With my new scheduler changes, a second `run()` call would fail because the 'flowstatus', ..., 'status' bit indicates the import step is "complete" (from the first run), causing the next task to launch too soon. I don't think there's a clean way to ignore this while also handling a partial steplist. In the meantime, I added a hack to that PR to clear these 'status' values after run.

Beyond this functional problem however, I think there's also a subtle issue here with regards to provenance. When you perform a second run(), new metrics and records will get stored side-by-side with the old ones. In most cases, old values will be totally overwritten, or the step lists of each run() will be totally orthogonal, so there's no real issue. However, if there is some step/index overlap things will get muddled in terms of what comes from where.

I think it would be worthwhile to fix this up. I have a few ideas:

1) We already have the 'history' config, so we could just clear everything that is set during the run ('metrics, 'record', 'flowstatus') and force all future lookups to read from history based on job name. I wonder if we can even have `chip.get(...)` transparently read from history based on the current job name, although that seems potentially dangerous.
2) Go back to having 'metric', 'record', and 'flowstatus' have per-job keys. Makes it a little less typing to access old values, but I think this is a little redundant with the 'history' dict.

aolofsson, 2022-03-30:
Good points. Not sure I see the larger conceptual issue here. What is the test case where this came up?

  1. The scope field is controls copying parameters to history, so no information should get lost. 
  2. There should nothing left in the current state when starting a run(). I like the idea of clearing out the record/metrics/flowstatus at the beginning of a run. (the "reentrant" use of run() on remote might affect implementation?)
  3. The history reflects previous jobs, anything non history reflects the current state, so having the job key mixxed into some parameters and not others feels like a partial solution. The "scope" filed gives us per parameter granularity on controlling the history of the parameter.
  4. The use of case of running through partial step lists on the same job run (one step at a time or "resume") would require the flowstatus to be retained between runs for the same job. What is the use case here?
  5. How are new metrics recorded "side by side", wouldn't they be overwritten the old values on a per step/index basis?

nmoroze, 2022-03-30:
One of the test cases that broke with the new "flowstatus" scheme is: https://github.com/siliconcompiler/siliconcompiler/blob/main/examples/gcd/gcd_skywater.py. "Larger conceptual issue" might have been a bit extreme, what I meant by that is that while the "flowstatus" issue is functional, the broader issue with metrics/record is conceptual (and hasn't been a problem yet).

> The scope field is controls copying parameters to history, so no information should get lost.

Agreed, not worried about info getting lost, just confusion about where info comes from.

> I like the idea of clearing out the record/metrics/flowstatus at the beginning of a run.

I'm not sure if we can do this, since that makes it impossible to continue a run by reading in a previous manifest, or making a second run() call with a partial steplist.

> The history reflects previous jobs, anything non history reflects the current state, so having the job key mixxed into some parameters and not others feels like a partial solution. The "scope" filed gives us per parameter granularity on controlling the history of the parameter.

Agreed! 

> The use of case of running through partial step lists on the same job run (one step at a time or "resume") would require the flowstatus to be retained between runs for the same job. What is the use case here?

That's a good point, my hack would break that use case. The one I'm trying to fix (shown in the example linked above) is running a totally different flow on the same chip object.

> How are new metrics recorded "side by side", wouldn't they be overwritten the old values on a per step/index basis?

Yeah, but I'm not sure the step/index would be enough to disambiguate. For example, suppose we added some basic metrics to our Surelog based import, and a user runs asicflow on their design. Then they run another flow with a built-in"nop" for import. The "import" metrics would include the minimal runtime for nop, but also the metrics from Surelog. Probably not a huge deal in this case since someone could figure out what's going on, but illustrates the point that you might have step/index overlap between different flows, and different tools report different metrics so not everything might get overridden. 

Those are good points overall, I've realized now that naively clearing things before OR after run() is problematic since it breaks re-running something with a partial steplist using the same chip object. I'll do some brainstorming to come up with a better solution, it will probably have to involve some sort of job name check, to determine if we're still working on an existing run or something fresh.

aolofsson, 2022-03-31:
> I'm not sure if we can do this, since that makes it impossible to continue a run by reading in a previous manifest, or making a second run() call with a partial steplist.

If we can figure out the last step of the graph executed in the last run() (which should be possible by checking the existence of output .json files), then we "should" be able to safely decide if the run() state should be cleared or continued. The alternative would be to possible make it user controlled with something like a -resume flag.  Still, we should be mindful of the steplist and the use cases supported. There are plenty of built in limitations with the steplist. For example,  trying to run a step before all the previous dependencies have executed correctly.

> Then they run another flow with a built-in"nop" for import. 

I think we still have some work to do with the modular flowgraphs, but each step within a named flow is unique, so there is no rerunning of import steps.

nmoroze, 2022-04-08:
I'd say this is fixed by #979.

closed 2022-04-08 by nmoroze