siliconcompiler/siliconcompiler #995: I cannot run build.py in microwatt example

jarvis-lorry, 2022-04-20:
I've tried to run build.py in .../examples/microwatt and I get an error about the missing attribute 'load_target'.  It seems I'm missing part of the installation for SiliconCompiler, though I've pulled the latest version from the repository.  

python3 build.py 
Traceback (most recent call last):
  File "build.py", line 81, in <module>
    main()
  File "build.py", line 67, in main
    chip.load_target('freepdk45_demo')
AttributeError: 'Chip' object has no attribute 'load_target'

aolofsson, 2022-04-21:
I am guessing you have an old version of SC. Can you try the following command at shell? The github repo is 0.8.0 compatible, we will try to get another release out next week.

```sh
$ sc -version
```

jarvis-lorry, 2022-04-22:
You're right: I have version 0.4.1 of SC.  Could you tell me how I can update it?  I originally installed SiliconComplier by cloning the github repo.

jarvis-lorry, 2022-04-28:
Hi @aolofsson ,
It's taken me some time to install an application called SureLog that SC requires and to update my version of yosys.  Now I'm faced with updated the openroad suite as my version isn't recent enough for SC.  With the commands above, should SC also update openroad as part of the third-party tools or is preferable to install OpenROAD separately?

aolofsson, 2022-04-28:
Can you move this over to the dissussion section? This is really a support question, not an issue.
In general, to update dependencies, you would go into the third_party and update submodules per the README.

closed 2022-04-28 by aolofsson

jarvis-lorry, 2022-04-28:
Thank you for your help, @aolofsson .