siliconcompiler/siliconcompiler #1006: SC 0.8.0 cannot find ghdl

jarvis-lorry, 2022-04-29:
I cannot run build.py in the microwatt example because SiliconCompiler 0.8.0 cannot find ghdl module.

| INFO    | job0  | import  | 0  | Copying /home/jarvis/siliconcompiler/third_party/designs/microwatt/fpga/top-generic.vhdl to 'inputs' directory
| INFO    | job0  | import  | 0  | Copying /home/jarvis/siliconcompiler/third_party/designs/microwatt/dmi_dtm_dummy.vhdl to 'inputs' directory
| ERROR   | job0  | import  | 0  | Executable ghdl not found
| ERROR   | job0  | import  | 0  | Halting step 'import' index '0' due to errors.
Traceback (most recent call last):
  File "build.py", line 81, in <module>
    main()
  File "build.py", line 78, in main
    chip.run()
  File "/home/jarvis/siliconcompiler/siliconcompiler/core.py", line 4215, in run
    raise SiliconCompilerError('Run() failed, see previous errors.')
siliconcompiler.core.SiliconCompilerError: Run() failed, see previous errors.

SC version
:~/siliconcompiler$ sc -version
0.8.0

nmoroze, 2022-04-29:
SC isn't having trouble finding the ghdl.py module, that error indicates that you don't have the ghdl executable installed (or it's not on your PATH).

This script may be helpful for building ghdl from source if you're running Ubuntu: https://github.com/siliconcompiler/siliconcompiler/blob/main/setup/install-ghdl.sh. Otherwise, take a look at their docs for instructions: https://github.com/ghdl/ghdl#getting-ghdl.

closed 2022-04-29 by nmoroze

jarvis-lorry, 2022-05-02:
Thank you @nmoroze for your help.  I apologize for not noticing that the SiliconCompiler build includes setup scripts for all of the third-party tools.