siliconcompiler/siliconcompiler #1007: Reordering library keypaths to avoid mixing dynamic dicts with fixed keys

nmoroze, 2022-04-29:
I noticed a bit of an inconsistency with some of the library schema keypaths.

Most of them follow the format of `['library', <lib>, (some fixed item), ...]`, e.g.: 

- `['library', <lib>, 'type']`
- `['library', <lib>, 'arch']`
-  `['library', <lib>, 'nldm', <corner>, <filetype>]`

However, there are a few that break this trend by taking free-form dict keys right after `<lib>`, e.g.: 
- `['library', <lib>, <pdk>, 'stackup']`
- `['library', <lib>, <design>, 'testbench']`
- `['library', <lib>, <design>, 'waveform']`

I think this is problematic for a few reasons:
- Design feels inconsistent
- Things will break if there are name conflicts b/w pdk, design, or any of the fixed keys.
- The sub-keys under `<pdk>` and `<design>` end up in the same 'default' dictionary, which I think some subtly strange implications. For example, if you do `chip.set('library', 'nangate45', 'freepdk45', 'stackup', '10M')`, then `chip.get('library', 'nangate45', 'freepdk45', 'testbench')` is a valid keypath.

I think it may be on the near-term roadmap to rethink the things under `['library', <lib>, <design>, ...]`, but in the meantime I propose inverting these keypaths as follows:

- - `['library', <lib>, 'stackup', <pdk>]`
- `['library', <lib>, 'testbench', <design>]`
- `['library', <lib>, 'waveform', <design>]`
etc.

Thoughts?

aolofsson, 2022-05-03:
You are right! Let's make sure we address this in the schema refactor under way. The 3 that notified were not well scoped out, so it makes sense that they might have been missed.

nmoroze, 2022-05-19:
Fixed by #1023.

closed 2022-05-19 by nmoroze