siliconcompiler/siliconcompiler #1022: Running the quickstart guide

arlted, 2022-05-17:
Per your recommendation, i followed the quick start guide to run the "heartbeat" example. 
Initially,  got the following error after executing "python heartbeat.py" in python virtual environment.  

"ERROR   | job0    | syn          | 0   | File ../third_party/pdks/virtual/freepdk45/libs/nangate45/r1p0/lib/NangateOpenCellLibrary_typical.lib was not found"

It looked like the "third_party" directory was missing in the "venv/lib/python3.6/site-packages" folder.  Then, i copied the "third_party" directory over the ./site-packages" folder to keep things running (this was not part of the guide). 

After running syn, floorplan, placement and cts steps, the execution stopped after giving out the following error
"[ERROR STA-0406] global_route positional arguments not supported. Error: sc_route.tcl, 36 STA-0406"

Am I missing something in the guide?

nmoroze, 2022-05-20:
For the first problem, that's a good workaround! The intended solution is to clone the repo and then create an environment variable `SCPATH` that points to `<path-to-repo>/siliconcompiler`. Sorry this wasn't stated in the documentation, I'll add it and leave this issue open till then.

For the second problem, I imagine it's an OpenROAD version issue. Which one do you have installed? (`openroad -version`)

arlted, 2022-05-20:
> All active links contained in this email were disabled. Please verify the identity of the sender, and confirm the authenticity of all links contained within the message prior to copying and pasting the address to a Web browser.

Thanks!!  The openroad version that I have is "v2.0-1644-g438dbb4".

nmoroze, 2022-05-20:
Yep, that's it! The minimum version our scripts support is [v2.0-3394-gd19162e82](d19162e829fdc1f683870ee4057cea9046fb8aae), so you'll need to upgrade.

By the way, I'd also recommend upgrading to the most recent version of SC (which we just published), since it's been a while since we've made a public release and there have been many changes since (including automated version checking!)

`pip install --upgrade siliconcompiler`

arlted, 2022-05-20:
great thanks!

nmoroze, 2022-05-25:
Docs issue updated in #1037. Feel free to open an issue if any other trouble arises!

closed 2022-05-25 by nmoroze