siliconcompiler/siliconcompiler #1063: Source code debugging

aolofsson, 2022-06-24:
The current SC approach copies files into the import directory before starting work, making it tough to debug source code for linting/dv since the reference is not the original code, but some file inside the build structure. The approach we took so far was done to enable remote processing and clean encapsulation of the design for rtl2gds, but it makes correlating the errors back to the original source file cumbersome.

For any action (linting, formal, dv, codecov) this is going to be a pretty big issue that we need to address.

nmoroze, 2022-06-24:
This is a good point, thanks for flagging this! I've also gotten feedback that this approach is somewhat confusing when you're first ramping up with SC (especially since the copied files go into `outputs/`). 

```
$ ls -l build/gcd/job0/import/0/outputs/
gcd_0a998a746037a840df8d7f2133c57f4846e3405f.v
gcd_9fb1319e7d576ade53f9b17ec8e05417afa27bca.sdc
gcd.pkg.json
gcd.v
```

without any knowledge of SC's internals, you're probably confused what the source files and big hashes correspond to. It's not obvious that gcd.v is a preprocessed output from Surelog and that the ones with big file hashes are verbatim inputs.

A step to clean up this confusion would probably be to just copy sources into `import/0/inputs/` (or somewhere else entirely), but we still have the problem you describe.

nmoroze, 2022-06-30:
@wasimzeroasicaltaf also flagged this during a discussion with me as something that could possibly be confusing - I think it's worth fixing.

As a simple fix, how about we just ignore the file `copy` flag when we're not doing a remote run? The only wrinkle is that it might be nice to copy in inputs for archiving, but we could add some extra logic to `archive()` that performs the copy and rewrites the manifest before tarring up the build directory.

closed 2022-07-26 by aolofsson