siliconcompiler/siliconcompiler #1067: Unable to input GDS for signoff flow

SeyiAyoARL, 2022-06-28:
https://github.com/siliconcompiler/siliconcompiler/blob/2c5039c96fbdbe64874cbf012c2edec379f0ff43/siliconcompiler/flows/signoffflow.py#L8

Hi there, I've tried to run DRC, and the comment in the signoff flow file describing how to provide input GDS is not working - here's the error message that is received:

| INFO    | Loading flow 'signoffflow'
| ERROR   | Get keypath [read,gds,drc,0] does not exist.
| INFO    | job0  | -       | -  | Checking manifest before running.
| ERROR   | job0  | -       | -  | Check failed. See previous errors.

In looking at the documentation, it looks like 'read' is not a part of the schema.

Thanks!

nmoroze, 2022-06-29:
We've since simplified this, but the comment wasn't updated - sorry for the confusion!

The new way to drive this flow is by setting inputs like so:

```python
    chip.set('input', 'gds', gds_path)
    chip.set('input', 'netlist', vg_path)
```

You can see this file for a more up-to-date example of an RTL2GDS + signoff flow: https://github.com/siliconcompiler/siliconcompiler/blob/main/examples/gcd/gcd_skywater.py. 

I'll leave this issue open until I get around to fixing that comment.

SeyiAyoARL, 2022-06-29:
Got it - thanks!

PR #1070 (2022-07-01): Fix some problems with docs
This PR fixes a couple recent issues pointing out bugs in our docs.

Closes #1067, #1066
closed 2022-07-01 by aolofsson