siliconcompiler/siliconcompiler #1072: summary generates error

arlpetergadfort, 2022-07-01:
On RHEL7 with Python3.6 I get the following error if I have the summary printed:
```
Traceback (most recent call last):
  File "examples/gcd/gcd-intel.py", line 83, in <module>
    main(root='examples/gcd')
  File "examples/gcd/gcd-intel.py", line 80, in main
    chip.summary()
  File "/data/Software/siliconcompiler/0.9.1/siliconcompiler/core.py", line 3163, in summary
    img_data = img_data,
UnicodeEncodeError: 'ascii' codec can't encode characters in position 7961-7962: ordinal not in range(128)
```

nmoroze, 2022-07-01:
Interesting, I'll look into this. There should be a .png file in your build directory (will probably be something like `build/gcd/job0/gcd.png`) -- if you're comfortable/able to send that to noah@zeroasic.com, that would be helpful!

arlpetergadfort, 2022-07-01:
That file is not there either

nmoroze, 2022-07-01:
That's odd. Are you able to send the gcd-intel.py script? Are you using a custom flow and/or PDKs/libs? I think I'll need more information to work out what's going on here.

PR #1081 (2022-07-15): Hardcode UTF-8 encoding for HTML report
There's a non-ASCII character (emdash) in the Bootstrap CSS file that gets inlined into our HTML report: https://github.com/siliconcompiler/siliconcompiler/blob/main/siliconcompiler/templates/report/bootstrap.min.css. 

On systems where the default encoding is UTF-8, this isn't a problem. However, when the default encoding is something else, the `wf.write(...)` call may raise a `UnicodeEncodeError`. Hardcoding the encoding to UTF-8 should fix this problem.

Thanks to @arlpetergadfort for reporting and finding the root cause of this issue! Closes #1072.
closed 2022-07-15 by aolofsson