siliconcompiler/siliconcompiler #1074: Finish implementing schema or document unimplemented features

nmoroze, 2022-07-05:
I was going through the set of schema options that are core runtime dependent to see how many have yet to be implemented.

I think there aren't too many at this point, I've counted about 7:

- `['flowgraph', <flow>, <step>, <index>, 'valid']`
- `['option', 'skipstep']`
- `['option', 'entrypoint']`
- `['option', 'metricoff']`
- `['option', 'trace']`
  - Not 100% sure about this one, since I don't really understand what it's supposed to do. It might be meant to be used by tool drivers.
- `['option', 'msgevent']`
- `['option', 'msgcontact']`
  - I'm not even sure how we'd go about implementing the messaging these (seems like we'd need some sort of backend for this, might have to be a cloud only thing?) The docs do already say "Actual support for email and phone messages is platform dependent", but we might want to be more specific.

I think in the short term, we should either: 

1) add an `implemented` flag to the schema to document which params aren't implemented
2) implement (or remove) each. Besides the message ones, I think these should all be easy to implement quickly

aolofsson, 2022-07-06:
- Agree that we should add an ['option', 'unimplemented'] key to track unimplemented features in the SC software.

> ['option', 'trace']
-This was meant to be a global option for outputting traces/waveforms from a simulator. (analog, digital, emulator, etc).

> ['option', 'msgevent']
-This was implemented in the client/server in one implementation by Will. It's a common feature for long runs in CAD systems.

nmoroze, 2022-07-06:
> Agree that we should add an ['option', 'unimplemented'] key to track unimplemented features in the SC software.

I was thinking of adding an additional field per keypath (like notes, signature, etc.), but that could work too. Are you thinking it would be a list of keypaths? 

For the other two, sounds good. I'll want to use ['option', 'trace'] for our Verilator driver, actually!

gadfort, 2024-12-18:
All features in this list have been implemented or the parameter has been deprecated.

closed 2024-12-18 by gadfort