siliconcompiler/siliconcompiler #1103: Use .odb rather than .def between OpenROAD steps

maliberty, 2022-07-27:
There is information that can't be stored in DEF so OpenROAD is moving away from it as a persistence format.  For example congestion, routing guides, pin access, Verilog hierarchy, and power domains have no representation in DEF.  RTL-MP and UPF support are near term features that will require odb.

You should use read_db / write_db rather than read_def / write_def.  ORFS has already made the switch and can be used as a reference.

maliberty, 2022-07-27:
Note that write_def remains available for interfacing with other tools (eg KLayout) at the end of the OR flow.

WRansohoff, 2022-07-27:
Thank you for the suggestion and context - this explains the removal of the `-guide` parameter in newer versions of OpenROADs' detailed routing steps.

aolofsson, 2022-08-02:
@maliberty In principle, what features are deprecated in the flow that would preclude the use of a def based flow? If the -guide input is gone from the detailed router, does that mean there is no way to take in results from an external placer/global router? Since openroad is currently the only APR that supports odb, this seems like a step backwards?

I think ASCII readable hand off should always be an available fall back option (even if only applicable at older nodes).

QuantamHD, 2022-08-02:
@aolofsson The DEF writer is still fully featured, but it lacks the ability to store tool specific information.

It's still fully possible to read_def and route that with openroad. What we're advocating for is that when you're using openroad the odb should take precedent over DEF.

DEF has a number of missing features that make it non ideal for flow infrastructure, namely it has no hierarchy support.

QuantamHD, 2022-08-02:
I do believe that in the long term we should try to move all open source tools to odb. DEF will always have a place, but odb will allow tighter integration with the rest of the open source tools.

WRansohoff, 2022-08-02:
Thanks for the extra information. Is it safe to read and write both OpenDB and DEF files to gather all available information from both formats? Or do we risk getting duplicate data if we call `read_db` followed by `read_def`?

QuantamHD, 2022-08-02:
In general those commands are pretty much mutually exclusive. Write commands are idempotent, and def is always a subset of odb so there would never be a benefit to do so for a *single block*.

read_def has a few modes on it that allow you to read pins or floorplan information in addition to the db..

 OpenROAD assumes a single unique instance of a chip/block in its code. Which read_def and read_db  either instantiate or overwrite. This is something @maliberty and I are working on rectifying.

In short read_db and read_def tend to overwrite everything.

aolofsson, 2022-08-02:
Thanks for the clarifications @QuantamHD !

> I do believe that in the long term we should try to move all open source tools to odb.

This is a great goal, but I have found that it is EXTREMELY difficult to convince anyone to adopt a standard. Babel's tower is the norm.  II hope it will always be possible to export/import ascii information from openroad that would allow modularity with non open source commercial tools (which will likely never be odb compatible).

- Is there a spec for the extra information in odb that gets used by place, cts, route?
- Is there a way to export this information as a JSON file so that it can be translated to the appropriate commands in others tools to complement a def read in those tools?

QuantamHD, 2022-08-02:
> This is a great goal, but I have found that it is EXTREMELY difficult to convince anyone to adopt a standard

Agreed. I'm advocating for a "put your money where your mouth is approach". i.e. we do the work to add support into those tools, and add functionality that isn't possible with the existing formats.

> Is there a spec for the extra information in odb that gets used by place, cts, route?

There is not. I suspect unless a process is put in place fields will get added where convenient.

>Is there a way to export this information as a JSON file so that it can be translated to the appropriate commands in others tools to complement a def read in those tools?

There is no way to do that today, and -guides was one example where that kind of functionality was removed. If the need arises OpenROAD could be enhanced to do that, but I don't think there's a priority to implement that functionality within the current developer/funding group. Could always be added by someone in the open source community.

nmoroze, 2022-08-02:
@QuantamHD on the subject of standards I'm curious if you or anyone on the OR dev team has looked into merging odb into [CIRCT](https://github.com/llvm/circt)? 

I originally had the impression that CIRCT was solely focused on frontends, but I saw recently that one of their broad forward-looking goals is to expand to IRs that capture physical design info (source: https://youtu.be/ee01_yHjs9k?t=2651). This might be a big lift at the moment (and I'm not sure what their roadmap is), but from my outsider perspective it seems like there's some philosophical alignment there.  

(Sorry for the tangent -- this is pretty much irrelevant to the original issue, and further discussion might best be kept to another issue in the OR repo. I'm also chiming in on this as as a personal interest, not really with my official SC dev hat on :))

aolofsson, 2022-08-03:
@QuantamHD @maliberty Hopefully this doesn't come across the wrong way....but I feel like I owe it to openroad to tell you that I think it is a mistake to make the platform less open.  I do understand the limitations of DEF, but there are plenty of cases where DEF works just fine. As an open platform. I believe it should strive to support the following scenarios without forcing people to adopt the odb database approach.

- Taking the output of openroad results (def/gateleve/props/etc) as an input to complete the job in an external tool. For example, there are times when the openroad is useful up until detailed route, but a different (commercial) solution is needed for the last step (detailed route + chip finishing).

- Taking the outputs from other tools as inputs to openroad to fill potential OR capability gaps. (physical synthesis, etc)

- Integration/etsting with legacy (and prototype) academic placers/routers for historical reasons which all work with DEF (and guides?) Integrating with opendb may be a barrier for collaboration and benchmarking with these tools.

- An ASCII readable well documented standard (DEF, JSON) invites the community to collaborate (anyone with Python skills can contribute). A  complex performance optimzied datastructure with undocumented features/fields (per your comment) reduces the potential for contribution.

QuantamHD, 2022-08-03:
@aolofsson It does not come across the wrong way. Good things happen when the community gets involved.

Just to be clear no one is advocating for getting rid of DEF. I'm advocating for another option in the long term. ODB is just a C++ wrapper on DEF with a very well documented API. https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/odb/include/odb/db.h (The entirety of the public API is in that file)

From what I've seen, projects like [DFFRAM](https://github.com/AUCOHL/DFFRAM) have used ODB successfully to make complicated things, but options are important. I'm fully supportive of DEF as a format.

Importantly, what we are advocating for in the this ticket specifically is that using DEF as an intermediate-representation for OpenLANE is non-ideal. Every use case you listed is still possible today, and more specifically DEF is the primary mechanism by which Klayout and Magic generate GDSII. DEF is here to stay, but in my opinion it should be primarily used when **Design Exchange** between tools is needed.

I think users of OpenLANE would be quite sad if one day we supported hierarchical verilog, but then it got flattened just after floorplanning. As a flow maintainer myself ([bazel_rules_hdl](https://github.com/hdl/bazel_rules_hdl/blob/main/place_and_route/open_road.bzl#L213)) I have always used (read/write)_db in my flow. I would advocate you do the same in SiliconCompiler except in cases when you need interchange between tools.

aolofsson, 2022-08-03:
@QuantamHD Thank you for the context! I agree 100% that end-to-end openroad flows like openlane should use odb and we will make that change. A small nit...an open source header file is a necessary but insufficient condition for broad adoption in open source. It really should have some kind of auto-generated manual (sphinx, doxygen,,,). That header files is massive.

QuantamHD, 2022-08-03:
Agreed on the documentation front. There's a lot between where we are and productive usage. There's no immediate plans of change just a north star of software defined hardware.

rovinski, 2022-08-03:
I just wanted to add, it case it's unclear, that there has been no loss in functionality, just a change to the API.

Before, the ASCII `.guide` file was being read and parsed by the detailed router. This is why there was a `-guide` flag for `detailed_route`.

Now, the guides are loaded/stored directly in odb. There are corresponding `read_guides` and `write_guides` commands to read from or write to an ASCII `.guides` file. As far as I know, there is no plan to remove this functionality or change the format.

In summary:
| Old flow | New flow | New flow with same intermediates |
|-|-|-|
|`global_route -guide_file GUIDE_FILE`<br>`detailed_route -guide GUIDE_FILE`|`global_route`<br>`// guides stored directly in odb`<br>`detailed_route`|`global_route`<br>`write_guides GUIDE_FILE`<br>`// close and reload openroad`<br>`read_guides GUIDE_FILE`<br>`detailed_route`|

aolofsson, 2022-08-03:
@rovinski Thanks for confirming that read/write guides is available!

maliberty, 2022-08-03:
read_lef/def will remain but you will get limited functionality in some cases if you go that route.

maliberty, 2022-08-06:
@aolofsson  Let me expand on my thoughts.  I don't think we should take away to ability to work in industry standard formats where they exist.  However all proprietary tools also have proprietary formats as storing, loading, and parsing text is quite inefficient.   In OR case we are making our database public and all the data fully accessible.

There is no industry standard for guides and most tools don't generally expose them.  We have been using a contest format and still support it for that purpose.  UPF provides partial support for power domains but additional data is necessary to physical implement them.  We would have to invent another format for that as well.   Any debugging information would need another format.  Clusters/partitions can be jammed into DEF but it isn't that great.  Etc.

Having to read/write 5-10 different files between steps is possible but not great.   

ML people recoil in horror at all this and just want to query a db for the data.

gadfort, 2023-01-16:
Merged with #1200

closed 2023-01-16 by gadfort