siliconcompiler/siliconcompiler #1116: Error with multiple VHDL input files

suppamax, 2022-08-02:
If I run the microwatt example, I get this error
```
| ERROR   | job0  | import  | 0  | Required file keypath ['input', 'vhdl'] can't be resolved.
| ERROR   | job0  | import  | 0  | Fatal error in check_manifest()! See previous errors.
```
I noticed that in the `heartbeat_caravel_wrapper` example there is one `chip.set()` followed by other `chip.add()`, but that didn't seem to help.
Using only `chip.set()` doesn't work because only the file specified with the last command is used.

What is the correct syntax?

nmoroze, 2022-08-03:
I think the "can't be resolved" error might not indicate that the syntax is problematic, but rather that the file(s) cannot actually be found. How are you specifying the path? (relative, absolute, any env vars in it?)

suppamax, 2022-08-03:
I have the same impression, that the files cannot be found.
The strange thing is that this happens if I use `cpu.add()`; with `cpu.set()`I don't get this error messages.
The problem with `cpu.set()` is that I cannot specify multiple files.

You can reproduce the error using the `milliwatt` example.

nmoroze, 2022-08-05:
Hm, it seems to work for me. A couple questions just as a quick check:

- are you running the script with `examples/microwatt` as your working directory?
- have you cloned the microwatt submodule from `third_party/designs/`?

suppamax, 2022-08-07:
I didn't notice that `microwatt` had not been cloned.
And because of a typo in my design I didn't notice one path was wrong.

Sorry for this.

closed 2022-08-07 by suppamax