siliconcompiler/siliconcompiler #1117: Surelog fails to parase parameter with partial vector selcet

aolofsson, 2022-08-03:
There seems to be an issue with the surelog parser for partial parameter selections. It's a problem because we rely on surelog to help us bring in files to the next step. 

https://github.com/chipsalliance/Surelog/issues/3133

I have veriried that the design goes through yosys with verilator as the front end parser.

mithro, 2022-08-03:
FYI - @hzeller / @kgugala

aolofsson, 2022-08-04:
Problem solved by updating to surelog 1.35.  

@nmoroze @WRansohoff This is a show stopper for the code I am working on, can we update SC with the latest surelog?

nmoroze, 2022-08-04:
Yeah, I can prep something to bump the Surelog. You might have been hoping to avoid this, but SC will use a Surelog on your PATH instead of the bundled one if you want to build the new version yourself in the meantime.

aolofsson, 2022-08-04:
Yes, already did on my machine. Still, figured we should update since this is a showstopper for some designs.

nmoroze, 2022-08-04:
Yup, absolutely. Just wanted to make sure you weren't blocked :)

PR #1129 (2022-08-25): Bump Surelog version
Fixes #1117

This one was kind of a struggle to get working. For some reason the bundled Surelog only works w/ Linux if we add the `-nocache` option. I'll need to revisit that when I get a chance. I also think it might be worth revisiting our approach to bundling Surelog, since upgrading is a pretty high burden.
closed 2022-08-25 by aolofsson