siliconcompiler/siliconcompiler #1149: Finalize the use of UHDM as a Yosys entry point

suzizecat, 2022-09-20:
Hi !

I recently tried a run with of the `skywater130_demo ` target for a full SystemVerilog design.
My setup contains (in this order, if it have any importance) : 
```python
chip.set('input',"verilog", sources)             # define list of source files
# ...
chip.set('option','frontend', 'systemverilog')
# ...
chip.load_target('skywater130_demo')          # load predefined target
```

And it seems that there is a Surelog run from which a generated verilog file is extracted as shown in surelog tool handler (`tools/surelog/surelog.py`)  `post_process` function.
However, it looks like the yosys synthesis script (`sc_syn.tcl) is able to handle UHDM inputs while it seems that this is not handled by the yosys tool handler (`yosys.py`) 

This might very well relate to a previous discussion we had in #849 where it was indicated that Surelog was not deemed mature enough.

If this still is the case, you might want to use this issue as a reminder/FAQ.
Otherwise, it looks to me that two main points should be addressed for this flow to work properly:

- [ ] Send the proper UHDM database from surelog to yosys (probably fairly easy when you know how SC internals works which I haven't fully grasped)
- [ ] Allow the use of one (or more) modules for Yosys in the command line command.

About the second point, and if I remember correctly, UHDM support can be built as a module that can be dynamically loaded in yosys by using `yosys -m /path/to/uhdm-module.something` (Or directly `-m uhdm` given the proper setup.) Which, IMHO is more convenient than building a custom Yosys for this job.
It might be interesting to note that VHDL support can be provided to Yosys through the same mechanism with GHDL (though highly experimental IIRC).

nmoroze, 2022-10-03:
Thanks for opening this issue! Everything you've said is correct, and I think we're at a point where we could start experimenting with feeding UHDM directly into Yosys - we just haven't had time to revisit it. We'll keep this issue around to track it.

suzizecat, 2022-12-03:
Hi @nmoroze !
I fiddled around recently with yosys and surelog.

As of today, and with the current yosys UHDM integration (see https://github.com/antmicro/yosys-systemverilog ) you can basically use `read_systemverilog` and pass a bunch of SystemVerilog files.

Therefore, the only required steps would be to send all SV files as input to the Yosys step and find a way to pass them to the `sc_syn.tcl` script. 
However, I'm a bit lost on how to do so.
Could you have a look at it or give me some hindsight about this so I can try and have a better look ?

gadfort, 2024-06-22:
Closing stale issue. If yosys will natively support UHDM then this can be revisited.

closed 2024-06-22 by gadfort