siliconcompiler/siliconcompiler #1163: Keypath related error while running ./examples/gcd/run.sh

arlted, 2022-10-05:
Hi,

Encountered the following error while running the  "./examples/gcd/run.sh".  Am I missing a something here?

siliconcompiler.core.SiliconCompilerError: Invalid keypath: ('library', 'nangate45', 'model', 'timing', 'nldm', ['typical'])
Your Chip configuration may be missing a parameter which is expected by your build script.

Thanks,
Ted

nmoroze, 2023-01-03:
Hi Ted,

Sorry for the long delay getting back - is this still an issue? If so, what version of SC are you using (`sc -version`), and what commit of the Git repo do you have downloaded?

arlted, 2023-02-15:
Sorry for the delayed response. No problem. I am now able to run the script successfully after pulling the latest SC version and commit.  We can close this issue. Thanks!

closed 2023-02-15 by gadfort