siliconcompiler/siliconcompiler #1166: Subtle filepath resolution issues

nmoroze, 2022-10-20:
Our shared method for resolving paths to data files as well as Python setup modules has gotten us into a bit of trouble. For example, one type of issue we've seen could arise when you have a directory structure like this:

```
foo.v
siliconcompiler/
myproj/
  asic/
     impl/
        myproj.py
   foo.v
```

myproj.py:

```python
chip.set('input', 'verilog', '../../foo.v')
```

Due to using SC root as a search path, this Verilog file could match the one in the top-level rather than the one in the project.

I believe the search order makes the CWD take priority, so I'll have to double check the exact weird behavior I saw in practice. However, overall I think we should eventually move to get rid of our fancy SCPATH/scroot filepath resolution once we move towards importing setup modules through the standard Python import mechanism.

gadfort, 2023-01-16:
@nmoroze I believe you merged a change that change the order when searching, if addressed in that PR, please close

nmoroze, 2023-01-16:
Yep, #1198 should fix this. Thanks for bumping!

closed 2023-01-16 by nmoroze