siliconcompiler/siliconcompiler #1393: Error fetching remote results

ryancor, 2023-03-15:
```
| INFO    | job0  | import     | 0  | Job is still running (1992 seconds, step: (Currently waiting in the queue)).
| INFO    | job0  | import     | 0  | Remote job run completed! Fetching results...
| WARNING | job0  | import     | 0  | Could not fetch results.
Traceback (most recent call last):
  File "/Users/ryancor/Documents/ASIC/PCIe/run.py", line 13, in <module>
    main()
  File "/Users/ryancor/Documents/ASIC/PCIe/run.py", line 8, in main
    chip.run()
  File "/usr/local/lib/python3.10/site-packages/siliconcompiler/core.py", line 4313, in run
    fetch_results(self)
  File "/usr/local/lib/python3.10/site-packages/siliconcompiler/client.py", line 348, in fetch_results
    chip.error(f"Sorry, something went wrong and your job results could not be retrieved. (Response code: {results_code})", fatal=True)
  File "/usr/local/lib/python3.10/site-packages/siliconcompiler/core.py", line 5025, in error
    raise SiliconCompilerError(msg) from None
siliconcompiler.core.SiliconCompilerError: Sorry, something went wrong and your job results could not be retrieved. (Response code: 500)
```
I got this issue for reference ID #28f266a6f39b4f689b5896b102c874a2

Also I'm not sure but it seems after this latest update, things are running a lot slower using remote

WRansohoff, 2023-03-15:
Thank you for helping us test our public beta!

This job ran out of memory in the routing step; it appears to be too large for the machines which we are using to run the jobs. There is also a bit more overhead because we are running things in isolated containers now, to account for the lack of signup and authentication.

We will discuss whether we can increase the size of the machines which we use to run these jobs.

WRansohoff, 2023-03-16:
Okay, the server should return partial results for jobs which run out of resources now, in the form of a PDF report which will be displayed if a complete GDS screenshot can't be generated. The PDF should include a detailed table of metrics for the steps which did complete successfully, and give you some indication of where the failure occurred.

Sorry for the confusion, and thank you for bringing this issue to our attention. We will discuss the public beta machines' resource limitations internally, in light of the fact that your build was limited by them.

I would also be interested to see what you get if you try to re-run the same job now. If you are running the jobs in a headless environment, the PDF report would be located at `build/[design]/[job_name]/report.pdf`.

WRansohoff, 2023-03-18:
@ryancor We have increased the size of our remote machines for the v0.10 release, and I think that your design should be able to build successfully now. It may take 2-3 hours with a long `route0` step though, and it is probably approaching the limit of complexity that we can support without using even larger machines.

If you find yourself wanting to build even larger designs, you can perform [a local install of SiliconCompiler](https://docs.siliconcompiler.com/en/stable/user_guide/installation.html#install-from-github-repo-linux-macos) and [its core EDA tool dependencies](https://docs.siliconcompiler.com/en/stable/user_guide/installation.html#external-tools) ([Surelog](https://github.com/siliconcompiler/siliconcompiler/blob/main/setup/install-surelog.sh), [yosys](https://github.com/siliconcompiler/siliconcompiler/blob/main/setup/install-yosys.sh), [OpenROAD](https://github.com/siliconcompiler/siliconcompiler/blob/main/setup/install-openroad.sh), and [KLayout](https://github.com/siliconcompiler/siliconcompiler/blob/main/setup/install-klayout.sh)). Local builds will also provide you with a full GDS-II result, in addition to a chip screenshot and metrics. Once everything is installed, you can run jobs on your local machine by omitting the `-remote` CLI option or the `chip.set('option', 'remote', True)` Python parameter.

Once again, sorry for the inconvenience, and thank you for trying SiliconCompiler. We want this build framework to be easy and intuitive to use, so reports like this which highlight confusing error responses are extremely helpful!

ryancor, 2023-03-22:
Hi @WRansohoff thank you for the response! Out of curiosity, why did you take the ability for users to get GDS files out of remote runs? It's such an awesome feature rather than having to spend hours locally compiling SiliconCompiler, I'm rerunning my mips16 processor chip remotely to see if what I get back!

WRansohoff, 2023-03-22:
Honestly, we were concerned about how much outgoing bandwidth might be consumed by sending back a GDS file for every design; they can get fairly large. We're hoping that the public remote servers can help people to iterate on designs and get them to a point where they can be built on open-source tools/PDKs, with the image-based summaries acting as a high-level gauge for optimization.

It would be nice to go back to returning a complete result which could be opened in KLayout, though. We might be able to address our bandwidth concerns by using the more efficient Oasis format instead of GDS. Thank you for the feedback!

WRansohoff, 2023-03-22:
We do support MacOS, but I think that most of our setup/install scripts for the open-source EDA tools are only written for Ubuntu. You can find build instructions for the tools on the [Surelog](https://github.com/chipsalliance/Surelog), [yosys](https://github.com/YosysHQ/yosys), and [OpenROAD](https://github.com/The-OpenROAD-Project/OpenROAD) repositories. I think that [KLayout has a packaged build](https://www.klayout.de/build.html#downloads) available for MacOS.

@azaidy or @jldec may have more information on the build process for the EDA tools; I think they are using MacOS internally. Sorry that we don't have ready-to-go setup scripts for the tools on that platform.

ryancor, 2023-03-22:
also another issue i just ran into after re-installing
```
➜  MIPS16 sc mips_16.v -remote

 ____  _ _ _                    ____                      _ _
/ ___|(_) (_) ___ ___  _ __    / ___|___  _ __ ___  _ __ (_) | ___ _ __
\___ \| | | |/ __/ _ \| '_ \  | |   / _ \| '_ ` _ \| '_ \| | |/ _ \ '__|
 ___) | | | | (_| (_) | | | | | |__| (_) | | | | | | |_) | | |  __/ |
|____/|_|_|_|\___\___/|_| |_|  \____\___/|_| |_| |_| .__/|_|_|\___|_|
                                                   |_|

Authors: Andreas Olofsson, William Ransohoff, Noah Moroze, Zachary Yedidia
Version: 0.1.1

--------------------------------------------------------------------------------
| INFO    | job0    | ---          | -   | Setting commandline arguments
| INFO    | job0    | ---          | -   | Loading target 'asicflow_freepdk45'
| INFO    | job0    | ---          | -   | Loading function 'setup_flow' from module 'asicflow'
| INFO    | job0    | ---          | -   | Loading function 'setup_pdk' from module 'freepdk45'
| INFO    | job0    | ---          | -   | Operating in 'asic' mode
| INFO    | job0    | ---          | -   | Waiting for inputs...
| INFO    | job0    | ---          | -   | Collecting input sources
| INFO    | job0    | ---          | -   | Copying /Users/ryancor/Documents/ASIC/MIPS16/mips_16.v to 'inputs' directory
| INFO    | job0    | ---          | -   | Writing manifest to /Users/ryancor/Documents/ASIC/MIPS16/build/mips_16/job0/import/0/sc_manifest.tcl
| INFO    | job0    | ---          | -   | Running in /Users/ryancor/Documents/ASIC/MIPS16/build/mips_16/job0/import/0
| INFO    | job0    | ---          | -   | surelog -parse /Users/ryancor/Documents/ASIC/MIPS16/mips_16.v -top mips_16 +libext+.sv
Traceback (most recent call last):
  File "/usr/local/bin/sc", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/local/lib/python3.11/site-packages/siliconcompiler/apps/sc.py", line 64, in main
    chip.run()
  File "/usr/local/lib/python3.11/site-packages/siliconcompiler/core.py", line 3124, in run
    remote_preprocess(self)
  File "/usr/local/lib/python3.11/site-packages/siliconcompiler/client.py", line 66, in remote_preprocess
    chip._runtask(local_step, index, active, error)
  File "/usr/local/lib/python3.11/site-packages/siliconcompiler/core.py", line 2957, in _runtask
    proc = subprocess.Popen(cmdlist,
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/Cellar/python@3.11/3.11.2_1/Frameworks/Python.framework/Versions/3.11/lib/python3.11/subprocess.py", line 1024, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/local/Cellar/python@3.11/3.11.2_1/Frameworks/Python.framework/Versions/3.11/lib/python3.11/subprocess.py", line 1901, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'surelog'
```

then also 

```
➜  MIPS16 python3 run.py
Traceback (most recent call last):
  File "/Users/ryancor/Documents/ASIC/MIPS16/run.py", line 13, in <module>
    main()
  File "/Users/ryancor/Documents/ASIC/MIPS16/run.py", line 5, in main
    chip.set('input', 'rtl', 'verilog', 'mips_16.v')
  File "/usr/local/lib/python3.11/site-packages/siliconcompiler/core.py", line 864, in set
    return self._search(cfg, keypathstr, *all_args, field=field, mode='set', clobber=clobber)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/siliconcompiler/core.py", line 1104, in _search
    cfg[param] = copy.deepcopy(cfg['default'])
                               ~~~^^^^^^^^^^^
KeyError: 'default'
```

my run.py script
```
import siliconcompiler

def main():
    chip = siliconcompiler.Chip('mips_16')
    chip.set('input', 'rtl', 'verilog', 'mips_16.v')
    chip.set('option', 'remote', True)
    chip.load_target('freepdk45_demo')
    chip.run()
    chip.summary()
    chip.show()

if __name__ == '__main__':
    main()
```

ryancor, 2023-03-22:
```
➜  MIPS16 brew install bison
==> Downloading https://formulae.brew.sh/api/formula.jws.json
##O#-#
==> Downloading https://formulae.brew.sh/api/cask.jws.json
##O=#  #
Warning: bison 3.8.2 is already installed and up-to-date.
To reinstall 3.8.2, run:
  brew reinstall bison
➜  MIPS16 brew install flex
Warning: flex 2.6.4_2 is already installed and up-to-date.
To reinstall 2.6.4_2, run:
  brew reinstall flex
  ```
  
got bison and flex

WRansohoff, 2023-03-22:
Looks like you might be missing the `surelog` tool executable.

[Surelog](https://github.com/chipsalliance/Surelog) is the tool which we use for linting and collecting HDL sources in the initial `import` stage, and we package it with the Python wheel that gets installed when you run `pip install siliconcompiler` because it is required for the remote workflow. If you install SiliconCompiler from this repository with something like `pip install -e .`, you'll also need to have a compatible version of Surelog installed on your system. Sorry for the confusion.

ryancor, 2023-03-22:
thats what super weird, i have it installed
```
➜  MIPS16 python3 -c "import siliconcompiler;print(siliconcompiler.__version__)"
0.1.1
```

it just like magically stopped working. i removed it locally from mac, and just installed via pip and just kept giving me this error

jldec, 2023-03-22:
@ryancor that looks like you might have python v3.11 which isn't currently supported by our pubished wheels.

(some of the very early pip releases of SiliconCompiler were less explicit about the python version so that's why your install may have picked up that very early build)

ryancor, 2023-03-22:
ahhh okay, my brew update might have screwed that up. I'm going to compile this locally on an ubuntu machine and let you know how it goes!

WRansohoff, 2023-03-22:
Hm, could you have moved the install script from its default location in the repository before running it?

It refers to another file in the repository which tracks which commit hash / version number should be installed ([`_tools.py`](https://github.com/siliconcompiler/siliconcompiler/blob/735408a509f4d039550224e7d1f2805aa4b91592/setup/install-surelog.sh#L15)), and uses a relative file path based on the script's location.

ryancor, 2023-03-22:
oh do the scripts need to be placed into the silicon compiler main dir?

ryancor, 2023-03-22:
oh i see, its in the setup directory!

WRansohoff, 2023-03-22:
The scripts are intended to be located in the `setup/` directory, but you should be able to run them from any location - something like `../../siliconcompiler/setup/install-surelog.sh` should work as well as running it from inside the repository. The setup scripts should install the tool system-wide, and I believe the build artifacts will be located in a `deps/` sub-directory wherever you ran the script.

Sorry for the confusion!

ryancor, 2023-03-22:
No problem! I apologize for all the messages haha I just really want to finish my MIPS16 asic :D

WRansohoff, 2023-03-22:
No problem at all, we're excited to see people using the tool with their own designs! The feedback is very helpful; we are trying to polish the user-facing messages and interfaces.

Good luck, we'd be interested to see what you end up with!

ryancor, 2023-03-22:
```
$ ls build/mips_16/job0/export/0/outputs/
ryancor@ubuntu:~/Desktop/Digital-Designs/MIPS16$
```
empty

ryancor, 2023-03-22:
```
ls build/mips_16/job0/export/1/outputs/
mips_16.def  mips_16.lef  mips_16.odb  mips_16.pkg.json  mips_16.sdc  mips_16.typical.lib  mips_16.typical.spef  mips_16.vg
```
no GDS in here

WRansohoff, 2023-03-22:
Can you confirm that KLayout is installed (`klayout -zz -v`)? And were there any logging statements starting like this prior to the summary (with `INFO`/`WARNING`/`ERROR`/etc)?

    | WARNING | job0  | export     | 0  | [...]

ryancor, 2023-03-22:
```
$ klayout -zz -v
KLayout 0.26.2
```

and i used another verilog sample that worked before with remote option
```
 python3 run.py 
| INFO    | job0  | --         | -  | Checking manifest before running.
| INFO    | job0  | import     | 0  | Tool 'surelog' found with version '1.40' in directory '/usr/local/bin'
| INFO    | job0  | import     | 0  | Running in /home/ryancor/Desktop/Digital-Designs/Pulse-Merge/build/pulse_merge/job0/import/0
| INFO    | job0  | import     | 0  | surelog -nocache -parse -nouhdm /home/ryancor/Desktop/Digital-Designs/Pulse-Merge/pulse_merge.v -top pulse_merge +libext+.sv+.v
[INF:CM0023] Creating log file ./slpp_all/surelog.log.

[WRN:PA0205] /home/ryancor/Desktop/Digital-Designs/Pulse-Merge/pulse_merge.v:1:1: No timescale set for "pulse_merge".

[INF:CP0300] Compilation...

[INF:CP0303] /home/ryancor/Desktop/Digital-Designs/Pulse-Merge/pulse_merge.v:1:1: Compile module "work@pulse_merge".

[INF:CP0302] Compile class "work@mailbox".

[INF:CP0302] Compile class "work@process".

[INF:CP0302] Compile class "work@semaphore".

[NTE:CP0309] /home/ryancor/Desktop/Digital-Designs/Pulse-Merge/pulse_merge.v:5:20: Implicit port type (wire) for "count_out",
there are 1 more instances of this message.

[INF:EL0526] Design Elaboration...

[NTE:EL0503] /home/ryancor/Desktop/Digital-Designs/Pulse-Merge/pulse_merge.v:1:1: Top level module "work@pulse_merge".

[NTE:EL0508] Nb Top level modules: 1.

[NTE:EL0509] Max instance depth: 1.

[NTE:EL0510] Nb instances: 1.

[NTE:EL0511] Nb leaf instances: 0.

[INF:UH0706] Creating UHDM Model...

[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 1
[   NOTE] : 6
| WARNING | job0  | import     | 0  | [WRN:PA0205] /home/ryancor/Desktop/Digital-Designs/Pulse-Merge/pulse_merge.v:1:1: No timescale set for "pulse_merge".
| INFO    | job0  | import     | 0  | Finished task in 0.68s
Opening file for replace: /home/ryancor/Desktop/siliconcompiler/third_party/pdks/skywater/skywater130/libs/sky130hd/v0_0_2/lib/sky130_fd_sc_hd__ss_n40C_1v40.lib.gz
Marked 40 cells as dont_use
Commented 0 lines containing "original_pin"
Replaced malformed functions 0
Replaced capacitive load 0
Writing replaced file: inputs/sc_dff_library.lib
Opening file for replace: /home/ryancor/Desktop/siliconcompiler/third_party/pdks/skywater/skywater130/libs/sky130hd/v0_0_2/lib/sky130_fd_sc_hd__ss_n40C_1v40.lib.gz
Marked 40 cells as dont_use
Commented 0 lines containing "original_pin"
Replaced malformed functions 0
Replaced capacitive load 0
Writing replaced file: inputs/sc_logiclib_sky130hd_sky130_fd_sc_hd__ss_n40C_1v40.lib.lib
```

this also finished as well no errors but no GDS

WRansohoff, 2023-03-22:
Ah, [you may need a newer version of KLayout](https://github.com/siliconcompiler/siliconcompiler/blob/735408a509f4d039550224e7d1f2805aa4b91592/siliconcompiler/tools/klayout/klayout.py#L78). There should be MacOS builds up to 0.28.x on [the KLayout download site](https://www.klayout.de/build.html#downloads).

We should probably print a clearer error message about that, but since the `export` step contains two parallel tasks, the `export0` failure may not have been very noticeable in the printed output.

ryancor, 2023-03-23:
hell yea! that fixed it! thank you so much. Check it out -> https://ibb.co/YysJpQt

ok so i guess the last big question i have is how do you add those pin pads? is there an easy way to do that in the python script?

WRansohoff, 2023-03-23:
Awesome! The padring generator is a little bit involved, but we do have an example which demonstrates how it can be used:

https://github.com/siliconcompiler/zerosoc

You should be able to build the example by running `python build.py --top-flat [--remote]`. We're hoping to add a tutorial for that design in the docs soon.

ryancor, 2023-03-23:
Yea I was looking at that but was a bit confused on which function was just solely responsible for the padgen. A tutorial for just that from scratch would be awesome. 

Oh btw I don’t know if you want to add this but using an ubuntu VM I had to allocate like 8 processor cores and 16Gbs of RAM. Yosys especially kept crashing the VM while compiling

gadfort, 2023-05-05:
@ryancor is your issue resolved?

gadfort, 2023-06-10:
Closing this due to inactivity, please reopen if closed prematurely

closed 2023-06-10 by gadfort

ryancor, 2023-06-14:
yes sorry @gadfort working great!