siliconcompiler/siliconcompiler #1451: Executable yosys not found

AHMEDALLAM86, 2023-04-07:
Hi ,

I am facing this error while running synthesis. Any solution for this?

Thanks
Ahmed

gadfort, 2023-04-07:
You'll need to have yosys installed along with the other tools needed for the flow (probably OpenROAD and klayout)
We have some helper scripts for Ubuntu 20: 

- https://github.com/siliconcompiler/siliconcompiler/blob/main/setup/install-yosys.sh
- https://github.com/siliconcompiler/siliconcompiler/blob/main/setup/install-openroad.sh

AHMEDALLAM86, 2023-04-07:
Thanks for your help. Anyway to install it on windows?

gadfort, 2023-04-07:
I use windows subsystem for Linux to have Linux applications running on my computer: https://learn.microsoft.com/en-us/windows/wsl/install

Beyond that I don't know if Yosys is supported in Windows

gadfort, 2023-05-05:
@AHMEDALLAM86 have you been able to resolve this?

gadfort, 2023-06-10:
Closing this due to inactivity, please reopen if closed prematurely

closed 2023-06-10 by gadfort