siliconcompiler/siliconcompiler #1505: Could not find target

techstuds, 2023-04-26:
Hi, 
I tried to setup a new pdk (by creating a new target at first) but facing this error when I run the heartbeat-example with the new target:
    siliconcompiler.core.SiliconCompilerError: Could not find target pdk_demo

pdk_demo.py is the target which I created in siliconcompiler/siliconcompiler/targets

Any solution for this? 

Thanks
Nils

gadfort, 2023-04-26:
@techstuds what version of siliconcompiler are you using? If there is a syntax error in pdk_demo it might fail to import.
In the most recent version 0.11.0 you can do:
```
from siliconcompiler.targets import pdk_demo

.....

chip.load_target(pdk_demo)
```

techstuds, 2023-04-27:
Thank you @gadfort, this worked (version 0.11.0).

closed 2023-04-27 by techstuds