siliconcompiler/siliconcompiler #1534: Routing congestion too high

techstuds, 2023-05-05:
I'm facing this error during routing with openroad, the full message is:
`[ERROR GRT-0119] Routing congestion too high. Check the congestion heatmap in the GUI and load reports/heartbeat_congestion.rpt in the DRC viewer.`
I attached the results of the DRC viewer here as like as the heartbeat_congestion.rpt file. 
Does anyone know what to do at this point?

Thanks!

[image]
[heartbeat_congestion.rpt.txt](https://github.com/siliconcompiler/siliconcompiler/files/11404365/heartbeat_congestion.rpt.txt)

gadfort, 2023-05-05:
In the report it looks like you only have 2 routing tracks available. So either you are applying a rather aggressive adjustment to the global router or there is something wrong with the way the tracks were generated. When the gcells are computed, they should have about 15 routing tracks in them (if I recall correctly).

You can turn on the routing tracks in the gui to view them, there is also a routing congestion heatmap that allows you to see the congestion by layer.

techstuds, 2023-05-08:
What could be wrong with the way the tracks were generated?

If I turn on the tracks in the gui I get this view. Is there anything helpful? 
[image]

The routing congestion heatmap failed because it has not been populated with data. What can I do?

gadfort, 2023-05-08:
There are a couple of things you can look at. In the build directory for this step there is a `replay.sh` file, you can edit the file to remove the `-exit` argument and replace with `-gui` that way when the routing fails the heatmap will have the data available to view.
The routing grid looks correct, so that makes me wonder about the settings. Specifically the min and max routing layers and what you set the layer adjustment factors to. All this is available in the manifest under tool/openroad/task/route/var/xx_adjustment. @maliberty do you have any suggestions?

techstuds, 2023-05-10:
For the heatmap this worked well:
[image]
As expected the routing congestion happen where the cells are located, so there is definitely a problem with the routing tracks in my mind. What could cause the low capacity of routing tracks (see heartbeat_congestion.rpt) ? 

I couldn't find the manifest or rather a path /task/route/var/xx_adjustment but I don't changed the settings for the routing layers and adjustment factors, so it's still MET1 to MET5 and 0.3 for all layers.

vijayank88, 2023-05-11:
@techstuds 
What is your placement density?
Is possible to attach a test case?

gadfort, 2023-05-11:
@techstuds could you attach the logs from the floorplan and route steps? They should be located in build/heartbeat/job0/floorplan/0/floorplan.log and similar for route?

techstuds, 2023-05-11:
Actually not. That's confusing me too.

gadfort, 2023-05-11:
> Something weird is happening here as your derated resources exceed your original resources:
> 
> ```
> [INFO GRT-0053] Routing resources analysis:
>           Routing      Original      Derated      Resource
> Layer     Direction    Resources     Resources    Reduction (%)
> ---------------------------------------------------------------
> MET1       Horizontal        242           250          -3.31%
> MET2       Vertical          242           358          -47.93%
> MET3       Horizontal        242           358          -47.93%
> MET4       Vertical          242           360          -48.76%
> MET5       Horizontal        242           360          -48.76%
> ---------------------------------------------------------------
> ```
> 
> Are you using a negative derate?

I've seen negative derates a few times, but never more than what could be explained by a rounding error.
Looks like GRT does enforce the numbers be positive: https://github.com/The-OpenROAD-Project/OpenROAD/blob/a9cdf12fc1bf543a515a33ea637a5996f20c5be4/src/grt/src/GlobalRouter.tcl#L49

@techstuds could you confirm that the MET1...MET5 are all routing layers in the LEF?

gadfort, 2023-05-11:
@maliberty I don't think there is a power grid, since the log doesn't appear to have anything indicating it.
I've added a print statement for the routing adjustments so it's clear:
https://github.com/siliconcompiler/siliconcompiler/pull/1546

techstuds, 2023-05-23:
@maliberty Are you talking about the "Openroad global routing grid derating"? I set this value to 0.3 for all layers. In fact, I don't even know what that is for, so I'm asking you to explain? I would also like to ask what derating is needed for and whether the values for original/derated resources ​​also have units?

Since I worked without a powergrid in the previous tests, I have now added this one. I've also used all routing layers from the LEF (another layer named METTPL) but negative derates still remains (except for METTPL). I added the logs of my last run: 

[floorplan.log](https://github.com/siliconcompiler/siliconcompiler/files/11542722/floorplan.log)
[route.log](https://github.com/siliconcompiler/siliconcompiler/files/11542723/route.log)
[heartbeat_congestion.rpt.txt](https://github.com/siliconcompiler/siliconcompiler/files/11542724/heartbeat_congestion.rpt.txt)

Now it looks like that:

[image]

gadfort, 2023-06-10:
@techstuds have you been able to resolve this issue?

techstuds, 2023-07-14:
@gadfort yes I was able to solve this issue. The negative derates were caused by the line-2-Via Pitch being much too large. This was related to the SPACING expressions of the metal layers in the technology LEF. After I changed these to SPACINGTABLES the routing basically worked, but for some reason only for the clocktree. I will open a new issue with it.

closed 2023-07-14 by techstuds