siliconcompiler/siliconcompiler #1547: Unable to use 'sc' in windows - Powershell

shariethernet, 2023-05-11:
`sc` is a keyword that stands for 'Set-Content` in Windows PowerShell

gadfort, 2023-05-11:
@shariethernet that's an unfortunate collision.
It looks like you can run: `Del -Force alias:sc` or `Remove-Alias -Force sc` to remove that alias from your current session of PowerShell. That should expose the installed siliconcompiler (assuming your path has it in there).
You can run `Get-Command -Type All sc` to see all the commands for sc and the in the order they will be selected.
In a virtual environment I was able to use sc in powershell after I removed the alias.

shariethernet, 2023-06-07:
Thanks. That solves the issue. However, sc also is responsible to interact with Windows Service Controller, so an option of offering silicon compiler's sc to be called by another alias would be great.

closed 2023-06-07 by shariethernet