siliconcompiler/siliconcompiler #1671: RePlace divergence detected error

ryancor, 2023-06-14:
```
[NesterovSolve] Iter: 700 overflow: 0.583543 HPWL: 956226
[NesterovSolve] Iter: 710 overflow: 0.579076 HPWL: 958131
[NesterovSolve] Iter: 720 overflow: 0.569644 HPWL: 965507
[NesterovSolve] Iter: 730 overflow: 0.5577 HPWL: 974746
[NesterovSolve] Iter: 740 overflow: 0.548687 HPWL: 981041
[NesterovSolve] Iter: 750 overflow: 0.542512 HPWL: 986448
[NesterovSolve] Iter: 760 overflow: 0.5396 HPWL: 1001737
[NesterovSolve] Iter: 770 overflow: 0.538572 HPWL: 1016441
[NesterovSolve] Iter: 780 overflow: 0.537715 HPWL: 1027078
[NesterovSolve] Iter: 790 overflow: 0.537715 HPWL: 1033658
[NesterovSolve] Iter: 800 overflow: 0.537715 HPWL: 1040998
[NesterovSolve] Iter: 810 overflow: 0.537715 HPWL: 1051940
[NesterovSolve] Iter: 820 overflow: 0.537715 HPWL: 1061126
[NesterovSolve] Iter: 830 overflow: 0.537715 HPWL: 1069578
[NesterovSolve] Iter: 840 overflow: 0.537715 HPWL: 1073897
[NesterovSolve] Iter: 850 overflow: 0.537715 HPWL: 1077300
[NesterovSolve] Iter: 860 overflow: 0.537715 HPWL: 1081346
[NesterovSolve] Iter: 870 overflow: 0.537715 HPWL: 1076083
[NesterovSolve] Iter: 880 overflow: 0.537715 HPWL: 1072078
[NesterovSolve] Iter: 890 overflow: 0.537715 HPWL: 1065461
[NesterovSolve] Iter: 900 overflow: 0.532552 HPWL: 1050625
[NesterovSolve] Iter: 910 overflow: 0.502432 HPWL: 1042669
[NesterovSolve] Iter: 920 overflow: 0.44268 HPWL: 1025777
[NesterovSolve] Iter: 930 overflow: 0.442549 HPWL: 1029223
[NesterovSolve] Iter: 940 overflow: 0.473972 HPWL: 1004714
[NesterovSolve] Iter: 950 overflow: 0.458905 HPWL: 997280
[NesterovSolve] Iter: 960 overflow: 0.443377 HPWL: 1009753
[NesterovSolve] Iter: 970 overflow: 0.380633 HPWL: 1019671
[NesterovSolve] Iter: 980 overflow: 0.367137 HPWL: 1018329
[ERROR GPL-0307] RePlAce divergence detected. Re-run with a smaller max_phi_cof value.
Error: sc_place.tcl, 17 GPL-0307
| WARNING | job0  | place      | 0  | Command failed with code 1. See log file /home/ryancor/Desktop/Digital-Designs/XOR_String/build/xor_string_encryption/job0/place/0/place.log
| ERROR   | job0  | place      | 0  | Halting step 'place' index '0' due to errors.
| ERROR   | job0  | cts        | 0  | Halting step due to previous error in place0
| ERROR   | job0  | cts        | 0  | Halting step 'cts' index '0' due to errors.
| ERROR   | job0  | route      | 0  | Halting step due to previous error in cts0
| ERROR   | job0  | route      | 0  | Halting step 'route' index '0' due to errors.
| ERROR   | job0  | dfm        | 0  | Halting step due to previous error in route0
| ERROR   | job0  | dfm        | 0  | Halting step 'dfm' index '0' due to errors.
| ERROR   | job0  | export     | 0  | Halting step due to previous error in dfm0
| ERROR   | job0  | export     | 0  | Halting step 'export' index '0' due to errors.
| ERROR   | job0  | export     | 1  | Halting step due to previous error in dfm0
| ERROR   | job0  | export     | 1  | Halting step 'export' index '1' due to errors.
Traceback (most recent call last):
  File "build.py", line 13, in <module>
    main()
  File "build.py", line 8, in main
    chip.run()
  File "/home/ryancor/Desktop/siliconcompiler/siliconcompiler/core.py", line 4463, in run
    self.error('Run() failed, see previous errors.', fatal=True)
  File "/home/ryancor/Desktop/siliconcompiler/siliconcompiler/core.py", line 5025, in error
    raise SiliconCompilerError(msg) from None
siliconcompiler.core.SiliconCompilerError: Run() failed, see previous errors.
```

i've gotten this error before with smaller verilog files but now it seems to be pretty persistent, dont know why I'm getting this error. Here is my verilog file in test along with my build.py
```
import siliconcompiler

def main():
    chip = siliconcompiler.Chip('xor_string_encryption')
    chip.set('input', 'rtl', 'verilog', 'xor_string_encryption.v')
    chip.clock('clk', period=10)
    chip.load_target('freepdk45_demo')
    chip.run()
    chip.summary()
    chip.show()

if __name__ == '__main__':
    main()
```

```
module xor_string_encryption(
  input clk,
  input rst,
  input [7:0] data_in,
  output reg [7:0] data_out
);

  reg [7:0] key;

  always @(posedge clk) begin
    if(rst) begin
      key <= 8'h0;
    end
    else begin
      key <= 8'b10101010;
    end
  end

  always @(posedge clk) begin
    if(rst) begin
      data_out <= 8'h0;
    end
    else begin
      data_out <= data_in ^ key;
    end
  end

endmodule
```

gadfort, 2023-06-14:
@ryancor could you run `sc-issue -generate -cfg build/.../place/0/inputs/....pkg.json` and upload the result.
We are seeing similar issues, but have not isolated it yet.

gadfort, 2023-06-14:
Was this a remote or local run?

ryancor, 2023-06-14:
local run, my python script is above ^^

gadfort, 2023-06-14:
@ryancor I'll run your script as provided, but please provide the sc_issue tar ball next time since it makes it easier for me to replicate and share the testcase for debugging

ryancor, 2023-06-14:
i dont seem to have the sc-issue command :/

ryancor, 2023-06-14:
[xor_string_encryption.pkg.json.zip](https://github.com/siliconcompiler/siliconcompiler/files/11750207/xor_string_encryption.pkg.json.zip)

here is the pkg.json file though

gadfort, 2023-06-14:
@ryancor it looks like you are on an older version of SC. Based on the schema version probably about 3 months (it predates the `sc-issue` command which was added to allow users to generate reproduceable testcases).
I ran your example on my machine and didn't have any issues. I would recommend updating
`python3 -m pip install --upgrade siliconcompiler` and see if that resolves it.
Note that the minimum version of openroad changed since then so you may need to update that as well.

ryancor, 2023-06-14:
ok i just updated and its giving me surelog version errors saying im on 1.40 when i should be on >=1.51. Any easy way to update it?

ryancor, 2023-06-14:
oh and updating openroad

gadfort, 2023-06-14:
You may also want to make sure yosys is up to date as well, since that changed about 2 months ago.

ryancor, 2023-06-14:
whats the command to update surelog?

gadfort, 2023-06-14:
I believe you are on a mac, so I don't know (if you get your siliconcompiler from pypi we bundle surelog in that).

ryancor, 2023-06-14:
im on linux, and i used install-surelog.sh to get it, but i dont see anyplace on how to update it

ryancor, 2023-06-14:
maybe if i delete the surelog director in setup/deps dir, and re-run the install script

gadfort, 2023-06-14:
yeah, I would just rebuild, those scripts are not tested for rebuilding. The actual version is selected from the _tools.json file which should ensure the version you are getting matches the ones in our CI

ryancor, 2023-06-14:
ok lemme try a rebuild on some of these and get back to you on the RePlace error and see if everything worked out

ryancor, 2023-06-14:
```
| INFO    | job0  | import     | 0  | surelog -nocache -parse -nouhdm /home/ryancor/Desktop/Digital-Designs/XOR_String/xor_string_encryption.v -top xor_string_encryption +libext| ERROR   | job0  | syn        | 0  | Cannot resolve path ../third_party/pdks/skywater/skywater130/libs/sky130hd/v0_0_2/lib/sky130_fd_sc_hd__ss_n40C_1v40.lib.gz in required file
```
getting this error now when running build.py

gadfort, 2023-06-14:
You will need to set `export SCPATH=<git clone of sc>/siliconcompiler` otherwise it cannot find those files.
It's on my list for next month to start addressing this to make it easier to operate without `SCPATH`

ryancor, 2023-06-14:
awesome everything worked! thank you!

gadfort, 2023-06-14:
@ryancor no problem

gadfort, 2023-06-22:
@ryancor closing since this is resolved. please reopen if there are still outstanding issues

closed 2023-06-22 by gadfort