siliconcompiler/siliconcompiler #1713: Schema docs

aolofsson, 2023-06-28:
Th section below seems outdated, we don't convert all types  to strings on Jason export, right?

"Input arguments and return values of the set/get/add core methods are encoded as native Python types. The JSON format does not natively support all of these data types, so to ensure platform interoperability, all SC schema parameters are converted to strings before being exported to a json file. Additionally, note that the parameter value ‘None’ gets translated to the “null”, True gets translated to “true”, and False gets translated to “false” before JSON export."

gadfort, 2023-06-28:
@aolofsson yes that is a little outdated, although not completely wrong. We rely on the native JSON conversions for booleans and None, etc (which happen to map like that).
What page is this on?

aolofsson, 2023-06-28:
https://docs.siliconcompiler.com/en/stable/reference_manual/schema.html

This is the part that @nomoroize cleaned up with the new setget.  "The JSON
format does not natively support all of these data types, so to ensure
platform interoperability, all SC schema parameters are converted to
strings before being exported to a json file."

Small thing I know...

PR #1738 (2023-07-09): Update schema docs
Closes #1713.
closed 2023-07-09 by gadfort