siliconcompiler/siliconcompiler #1714: Post Process OpenRoad Floorplan

ryancor, 2023-06-28:
I got an issue with floor planning:
[image]

This is my verilog code
```
module det_1011(
	input clk,
	input rstn,
	input in,
	output out
);

	parameter IDLE = 0,
		  S1 = 1,
		  S10 = 2,
	    	  S101 = 3,
		  S1011 = 4;

	reg [2:0] cur_state, next_state;

	assign out = cur_state == S1011 ? 1 : 0;

	always @(posedge clk) begin
		if (!rstn) begin
			cur_state <= IDLE;
		end else begin
			cur_state <= next_state;
		end
	end

	always @(cur_state or in) begin
		case(cur_state)
			IDLE : begin
				if(in) begin
					next_state = S1;
				end else begin
					next_state = IDLE;
				end
			end

			S1 : begin
                                if(in) begin
                                        next_state = IDLE;
                                end else begin
                                        next_state = S10;
                                end
                        end

			S10 : begin
                                if(in) begin
                                        next_state = S101;
                                end else begin
                                        next_state = IDLE;
                                end
                        end

			S101 : begin
                                if(in) begin
                                        next_state = S1011;
                                end else begin
                                        next_state = IDLE;
                                end
                        end

			S1011 : begin
				next_state = IDLE;
                        end
		endcase
	end
endmodule
```

this is the build script:
```
import siliconcompiler

def main():
    chip = siliconcompiler.Chip('det_1011')
    chip.set('input', 'rtl', 'verilog', 'fsm.v')
    chip.clock('clk', period=10)
    chip.load_target('skywater130_demo')
    chip.run()
    chip.summary()
    chip.show()

if __name__ == '__main__':
    main()
```

ryancor, 2023-06-28:
got a bunch of issues when trying to generate with sc-issue 

[image]

ryancor, 2023-06-28:
Another thing, with verilog file that use to work before I updated it last time we spoke, now im getting this issue for every single build that worked before...

[image]

gadfort, 2023-06-29:
@ryancor this should have been fixed in: https://github.com/siliconcompiler/siliconcompiler/pull/1674 so if you update that will not be an issue.

The issue with sc-issue, is that you are attempting to package up the place0 node, instead of the failing floorplan0. I'll take a look to see if there is a better error message that can be provided.

ryancor, 2023-07-01:
[image]

@gadfort updated, now getting this error. Do i need to update yosys? im on version 0.27

gadfort, 2023-07-01:
@ryancor we switched the map_adders parameter in Yosys so this looks like you are running with an older version of the Yosys scripts. What commit hash or version of silicon compiler are you running?

ryancor, 2023-07-01:
im on the main branch, not sure about the commit hash but i did a pip upgrade and im on latest version of siliconcompiler. should i pull from main maybe?

ryancor, 2023-07-01:
[image]

so this should show you where im at in main

ryancor, 2023-07-01:
[image]
ok after pulling from main, got new errors

gadfort, 2023-07-01:
@ryancor you'll need to update OpenROAD, they fixed a typo in one of their commands

gadfort, 2023-07-01:
That should be the last change you need

ryancor, 2023-07-01:
doesn't the latest install-openroad.sh script install the appropriate version?

gadfort, 2023-07-01:
@ryancor yes it should.

ryancor, 2023-07-01:
[image]

awesome, all good again. Thanks for being super on top of bugs lol seems like you all catch them before i do :D

gadfort, 2023-07-05:
@ryancor great to hear.