siliconcompiler/siliconcompiler #1732: Mac dynamic library libc++.1.0.dylib needed by surelog

mayagokhale, 2023-07-07:
siliconcompiler install puts a pre-built surelog into the venv siliconcompiler tool surelog bin directory if an existing surelog binary is not found. on the Mac, this binary expects a Monterey libc++.1.0.dylib in /usr/lib. On other versions of MacOS (such as Big Sur), that library doesn't exist and surelog --version fails.
Luckily I had this library in a different location under anaconda, and adding the path to it to DYLD_LIBRARY_PATH solved the problem. 
I'm submitting this as an issue in case you want to add the fix information to the siliconcompiler build instructions.

nmoroze, 2023-07-20:
Hi, thanks for opening the issue! When you get a chance, could you please send the exact error message you get when running `surelog --version` without the updated library path?

mayagokhale, 2023-07-20:
venv/lib/python3.11/site-packages/siliconcompiler/tools/surelog/bin/surelog --version
dyld: Symbol not found: __ZTTNSt3__114basic_ifstreamIcNS_11char_traitsIcEEEE
  Referenced from: /Users/gokhale2/src/siliconcompiler/venv/lib/python3.11/site-packages/siliconcompiler/tools/surelog/bin/surelog (which was built for Mac OS X 12.6)
  Expected in: /usr/lib/libc++.1.dylib

nmoroze, 2023-07-20:
Thanks! I've attached a recompiled binary that I expect to work on Big Sur as-is, it would be helpful if you could un-gzip and try it out.

[surelog.gz](https://github.com/siliconcompiler/siliconcompiler/files/12114596/surelog.gz)

mayagokhale, 2023-07-20:
./surelog --version
VERSION: 1.69
BUILT  : Jul 20 2023

PR #1790 (2023-07-21): Ensure compiled Surelog is backwards compatible with older macOS versions
This PR fixes the bug that our compiled Surelog binary reports a missing system library on macOS Big Sur, even though the wheels are marked as compatible with this version. We need to define a special env variable `MACOSX_DEPLOYMENT_TARGET` that the macOS compiler uses to determine backwards compatibility requirements. This variable was previously set in the context where our leflib C-extension was built, but not for Surelog.

@mayagokhale tested out a binary built with this env var set, I'm re-running a wheels build fro [... cut]
closed 2023-07-21 by nmoroze