siliconcompiler/siliconcompiler #1789: /<identifier> being replaced by #~@<identifier#~@ which yosys doesn't like

mayagokhale, 2023-07-20:
"wire [1023:0] \mc_proc.bbufi_local_0 ;" gets re-written to "wire [1023:0] #~@mc_proc.bbufi_local_0#~@;"
Then 

yosys> read_verilog -sv inputs/zhw_encode_module.v

1. Executing Verilog-2005 frontend: inputs/zhw_encode_module.v
inputs/zhw_encode_module.v:22: ERROR: syntax error, unexpected '~'

gadfort, 2023-07-21:
@mayagokhale I believe I've seen this before when using the RTL has already been converted from SystemVerilog to Verilog by sv2v and then surelog does some reformatting.
Would you be able to share a testcase?

mayagokhale, 2023-07-21:
That is what I am seeing also: I isolated it to 'surelog -parse' changing those valid verilog identifier delimiters to the escape sequence. 

yosys is fine with original generated file (generated from SystemVerilog input). I used yosys 0.30 with the ant SystemVerilog plugin to process the SystemVerilog input file, ending with 'write_verilog' to output the version given to siliconcompiler. The version of yosys in siliconcompiler doesn't include the plugin, so I had to do that separately.

myfile.sv -> yosys+plugin -> myfile.v -> (siliconcompiler) surelog -> build/.../myfile.v

I'm trying to figure out how to modify the asicflow to skip surelog or change it to their linter. 

I don't have a small test case. The system verilog is 114,547 lines. It's generated from a SystemC program which started life at https://github.com/anikau31/systemc-clang/blob/fccm-ae/examples/llnl-examples/zfpsynth/shared3/zhw_encode.h

mayagokhale, 2023-07-22:
I "solved" this problem with a small modification to parse.py in tools/surelog. This isn't really a solution but allowed forward progress on my design, so in case anyone else is stuck on the surelog escape character problem, the diffs are attached.
[diffs.txt](https://github.com/siliconcompiler/siliconcompiler/files/12137173/diffs.txt)

gadfort, 2023-07-24:
@mayagokhale I see what you did. This is basically bypassing surelog (which works for your case).
With our current support, I would suggest using sv2v as the frontend instead of surelog.
To do so, just set `['option', 'frontend'] = 'systemverilog'`, that should ensure the conversion to verilog is handled by sv2v without the need for the preprocessing step you have.

I'd like to get the "long" testcase, since this is something that needs to be submitted as an issue to the Surelog project, if you don't mind? That way we can hopefully fix this in the future.

mayagokhale, 2023-07-25:
@gadfort thank you for the reply and the suggestion. I added that option (before the load_target as stated in the documentation), and reverted to the original parse.py in tools/surelog. However convert gave the same error.

cat build/encode_stream_sc_module_8/job0/convert/0/convert.log 
inputs/encode_stream_sc_module_8.v:517:15: Parse error: unexpected token '#' (Sym_pound)

A smaller test case that demonstrates the issue is attached. Please change the suffix from .txt to .v.
[encode_stream_sc_module_8_return_stmt_removed_yosys.txt](https://github.com/siliconcompiler/siliconcompiler/files/12164134/encode_stream_sc_module_8_return_stmt_removed_yosys.txt)

gadfort, 2023-07-26:
@mayagokhale the file you uploaded is not a systemverilog file, so I'm a bit confused about what the starting file looks like that you used yosys to convert from SV to V. If sv2v that would be useful to know as well.

If you could provide your source file and build script that would be tremendously helpful in debugging and improving the tool.

Either way, you can try this branch: https://github.com/siliconcompiler/siliconcompiler/tree/yosys-as-frontend
I bypassed surelog and added the code needed for yosys to read-in source files directly. I'd like to bring some of these changes in, but it needs more testing.

mayagokhale, 2023-07-26:
@gadfort  it has been a very confusing process working around the various tools' unique bugs to get to a design that made it through siliconcompiler. The file I provided was generated by yosys 0.30 using the write_verilog command. It gives an error in the convert process with and without  'frontend option verilog'. It demonstrates that surelog is re-writing identifiers '/<id> ' with the escape sequence, which then can't make it through yosys. 
(I couldn't use sv2v because the larger design has 2D array ports, which sv2v didn't like. I used the ant systemverilog plugin for that, but it has its own issues that I had to work around.)

Attached is a version before going through yosys, as you requested.

[encode_stream_sc_module_8.sv.txt](https://github.com/siliconcompiler/siliconcompiler/files/12166775/encode_stream_sc_module_8.sv.txt)

Thank you for the fixes in the new branch. I will try it out.

gadfort, 2023-07-26:
I opened two issues based on the problems you identified:
https://github.com/chipsalliance/Surelog/issues/3776
https://github.com/zachjs/sv2v/issues/249

Thanks

mayagokhale, 2023-07-26:
@gadfort I tried to install the yosys-as-frontend branch but there were compile errors in the openroad install probably due to my environment using clang rather than gcc. 
I also looked at temporarily replacing the 4 files in my current install but there were too many diffs in core.py between the new version and my installed core.py, so that would likely not be successful. I can work on installing openroad this weekend.

As far as how to make install easier, many people use dockerfiles. That's what we do in our tool and it simplifies the environment problem.

About problems with other tools, I also put in an issue at https://github.com/antmicro/yosys-systemverilog/issues, 1843. It causes yosys to abort on encode_stream when using their systemverilog plugin.

closed 2023-07-27 by nmoroze

gadfort, 2023-07-27:
@mayagokhale the solution that was merged in should allow you to correctly parse your verilog file.

PR #1805 (2023-07-27): surelog: undo escape character renaming
Closes #1789 

Per https://github.com/chipsalliance/Surelog/issues/3776 this adds a substitution to the files getting concatenated by Surelog.