siliconcompiler/siliconcompiler #1848: Getting started code does not work

AllenDBoston, 2023-08-09:
As of today after a fresh download of SC in a new pyenv the example code in the Getting Started portion of the ReadMe does not work.

gadfort, 2023-08-09:
@AllenDBoston could you elaborate on which portion of the getting started guide that doesn't work? And which version of SC you are using.

gadfort, 2023-08-09:
@AllenDBoston it appears to be a bug in the way python handles multiprocessing for spawn.
It's updated in: https://github.com/siliconcompiler/siliconcompiler/pull/1856

aolofsson, 2023-08-14:
I think you may have to remove the quotes around asic_demo.

AllenDBoston, 2023-08-16:
In fact the last line that includes "sc -remote -target "asic_demo"" works fine for me. 

Again, what I did was I downloaded SiC using a new pyenv then I run the following lines of python as given in the getting started readme:

import siliconcompiler                             # import python package
chip = siliconcompiler.Chip('heartbeat')           # create chip object
chip.load_target('skywater130_demo')               # load a pre-defined target
chip.input('heartbeat.v')                          # set input sources
chip.clock('clk', period=10)                       # set constraints
chip.set('option','remote', True)                  # enable remote execution
chip.run()                                         # run compilation
chip.summary()                                     # print summary
chip.show()                                        # show layout

In a new pyenv, on initial run of the code above I get that there is no module found for 'altair.vegalite.v4'. Therefore I fix with installing the correct version asked by SiC by executing "pip install altair==4".
 After that a new error related to multiprocessing spawn as mentioned above by @gadfort. As of today the getting started code still does not work. Thanks, Allen

gadfort, 2023-08-17:
@AllenDBoston we updated the instructions: https://docs.siliconcompiler.com/en/latest/user_guide/quickstart.html
The issue you are running into is how Python multiprocessing works with spawning. It will reload your original file, which causes all code to be re-executed and python does not like that. This is a known issue with spawning and the solution is to wrap your code as shown in https://docs.siliconcompiler.com/en/latest/user_guide/quickstart.html#id1

It's not 100% clear to be if you are using the docs or the github README.
What version of python are you using? (for the altair issue)

gadfort, 2023-08-17:
I've added an additional node to the github readme: https://github.com/siliconcompiler/siliconcompiler/pull/1876

gadfort, 2023-08-17:
@AllenDBoston I also added: https://github.com/siliconcompiler/siliconcompiler/pull/1880 thanks for the suggestion.

AllenDBoston, 2023-08-17:
Looks like the fix to include if __name__ == "__main__": fixed the issue with the getting started code on the GitHub readme.

gadfort, 2023-08-17:
@AllenDBoston thats great to hear, in that I'll close this. If you encounter any other issues, please feel free reopen this or file a new issue.