siliconcompiler/siliconcompiler #1893: ModuleNotFoundError: No module named 'six'

karbapi, 2023-08-20:
Hi,

First of thanks for giving such a prospective tool. However, I am having issue with the following:

```(base) ~ % python
Python 3.9.16 | packaged by conda-forge | (main, Feb  1 2023, 21:38:11) 
[Clang 14.0.6 ] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import siliconcompiler
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
....
....
ModuleNotFoundError: No module named 'six'```

Anything wrong with my installation? Please help me. Thanks.

gadfort, 2023-08-20:
That looks to be a requirement from `python-dateutil` which is used in pandas. Could you check that this is installed?
```
pip show python-dateutil
pip show six
```
This is something that should have been installed.
Maybe you can try to reinstall SC and copy in the log from that incase there are any warnings.

karbapi, 2023-08-21:
Thanks for your response. Below is the summary:

(base) ~ % pip show python-dateutil
Name: python-dateutil
Version: 2.8.2
Summary: Extensions to the standard Python datetime module
Home-page: https://github.com/dateutil/dateutil
Author: Gustavo Niemeyer
Author-email: gustavo@niemeyer.net
License: Dual License
Location: /Users/bapikar/miniforge3/lib/python3.9/site-packages
Requires: six
Required-by: matplotlib, pandas, streamlit
(base) ~ % pip show six
Name: six
Version: 1.15.0
Summary: Python 2 and 3 compatibility utilities
Home-page: https://github.com/benjaminp/six
Author: Benjamin Peterson
Author-email: benjamin@python.org
License: MIT
Location: /Users/bapikar/miniforge3/lib/python3.9/site-packages
Requires: 
Required-by: astunparse, google-auth, google-pasta, isodate, Keras-Preprocessing, python-dateutil, tensorflow-macos, tensorflow-metal
(base) ~ % pip show pandas
Name: pandas
Version: 1.4.3
Summary: Powerful data structures for data analysis, time series, and statistics
Home-page: https://pandas.pydata.org
Author: The Pandas Development Team
Author-email: pandas-dev@python.org
License: BSD-3-Clause
Location: /Users/bapikar/miniforge3/lib/python3.9/site-packages
Requires: numpy, python-dateutil, pytz
Required-by: altair, siliconcompiler, stellargraph, streamlit

gadfort, 2023-08-21:
@nmoroze any ideas? It looks like it's installed. Maybe it has something to do with conda?

karbapi, 2023-08-21:
just FYI, I have the latest conda installed (23.7.2)

karbapi, 2023-08-23:
It worked. Thanks a lot.

nmoroze, 2023-08-23:
Great! Is using SC outside of your conda environment an acceptable solution for you?

karbapi, 2023-08-24:
It’s okay. I will update if I encounter any further issues. Thanks again pal

nmoroze, 2023-08-24:
You're welcome!