siliconcompiler/siliconcompiler #1986: Could not find remote server configuration: defaulting to https://server.siliconcompiler.com

yyh-sjtu, 2023-09-22:
[image]
After I executed the command "sc -remote -target "asic_demo"", the shell told me that "Could not find remote server configuration" and the "https://server.siliconcompiler.com" cannot be visited either. 
[image]

gadfort, 2023-09-22:
That is just a notice that your job will be submitted to the default server (it does not have a public facing website, SC will upload the job to it and it will run it and return the results).
The failure you are encountering seems to relate to an issue with Surelog. What OS and python versions are you using? and did you install Surelog separately?

yyh-sjtu, 2023-09-22:
Thank you very much for reply. The OS is Windows, and the version of python is 3.10.0. I haven't install Surelog, because I don't know much about the siliconcompiler. I think it can run without any other tools, if I choose the remote option.

yyh-sjtu, 2023-09-22:
> That is just a notice that your job will be submitted to the default server (it does not have a public facing website, SC will upload the job to it and it will run it and return the results). The [...]

Thank you very much for reply. The OS is Windows, and the version of python is 3.10.0. I haven't install Surelog, because I don't know much about the siliconcompiler. I think it can run without any other tools, if I choose the remote option.

gadfort, 2023-09-22:
It looks like the surelog that we bundle with Windows did not compile properly. We'll look into that.

yyh-sjtu, 2023-09-22:
> It looks like the surelog that we bundle with Windows did not compile properly. We'll look into that.
OK. Thank you very much. You reply helps me a lot. I think I can choose linux OS to use siliconcompiler.

yyh-sjtu, 2023-09-22:
> It looks like the surelog that we bundle with Windows did not compile properly. We'll look into that.
OK. Thank you very much. Your reply helps me a lot. I think I can choose linux OS to use siliconcompiler.

PR #1994 (2023-09-25): build surelog with static zlib
Build surelog with static linking for zlib dependency: https://github.com/siliconcompiler/siliconcompiler/suites/16508799539/artifacts/943209333

Closes: #1986
closed 2023-09-25 by gadfort