siliconcompiler/siliconcompiler #2036: Error while running the asic demo

prathiksha97, 2023-10-11:
[image]
I am getting an error when I try to run the asic demo.
I am on Ubuntu 22.04 and the python version I am using is 3.10.13
Can anyone please help me with this

WRansohoff, 2023-10-11:
Hi there,

Sorry for the inconvenience. There is a bug in the client with v0.16.0 which causes the remote flow not to work. We should have a patch shortly.

If you are able to modify your installation, you should be able to work around the issue by replacing [the following two lines here](https://github.com/siliconcompiler/siliconcompiler/blob/7eaaaaba52daf418850233565b92a3f9f18045c2/siliconcompiler/remote/client.py#L173):

```
-     chip.set('arg', 'step', preset_step)
-     chip.set('arg', 'index', preset_index)
+    chip.set('option', 'from', 'syn')
+    chip.set('option', 'to', 'export')
```

We should have a more complete fix soon. Thank you for trying SiliconCompiler, and sorry again for the confusion.

closed 2023-10-11 by gadfort

gadfort, 2023-10-11:
@prathiksha97 @Vinayakamk a new release with the fix was pushed: https://github.com/siliconcompiler/siliconcompiler/releases/tag/v0.16.1

prathiksha97, 2023-10-18:
Thank you, It is working!

PR #2038 (2023-10-11): version 0.16.1
Closes #2036