siliconcompiler/siliconcompiler #2050: siliconcompiler._common.SiliconCompilerError: Server responded with 400: Error: Submitted jobs must use Schema version 0.37.x. Your Schema version is 0.36.0.

yyh-sjtu, 2023-10-17:
Dear developer:
I used to run the flow by remote server successfully. But today, when I run the same script. I find the problem:"siliconcompiler._common.SiliconCompilerError: Server responded with 400: Error: Submitted jobs must use Schema version 0.37.x. Your Schema version is 0.36.0."
[image]
How can I deal with it.

gadfort, 2023-10-17:
@yyh-sjtu you will need to update your SC version to atleast the 0.16.0 release.

yyh-sjtu, 2023-10-17:
> @yyh-sjtu you will need to update your SC version to atleast the 0.16.0 release.

OK, thank you very much. I will try.

closed 2023-10-19 by troibe