siliconcompiler/siliconcompiler #2056: Can siliconcompiler support "generate" literal in verilog?

yyh-sjtu, 2023-10-24:
[image]
[image]
[image]
The verilog code is tested. It can be compiled and can pass sythesis. But siliconcompiler reports error. Can siliconcompiler support "generate" literal in verilog?

gadfort, 2023-10-24:
Yes, we use them extensively ourselves.
Nothing special is needed to enable it, but it might be the connection is failing to download the results.

troibe, 2023-10-24:
Looking at your log there might also be an issue in the cts step.
It looks like cts and the following steps are marked as pending but then never get fetched.
Which can indicate there was a failure in cts.

troibe, 2023-10-24:
Siliconcompiler should download logs for each step so you can have a look at them in your build directory.
This might give you more hints.
If you want to view the logs directly in your console then that's unfortunately only possible if you run without the `remote` option.

yyh-sjtu, 2023-10-24:
> Yes, we use them extensively ourselves. Nothing special is needed to enable it, but it might be the connection is failing to download the results.

Thank you.

yyh-sjtu, 2023-10-24:
> Siliconcompiler should download logs for each step so you can have a look at them in your build directory. This might give you more hints. If you want to view the logs directly in your console then that's unfortunately only possible if you run without the `remote` option.

OK, thank you for reply. I will check it.

gadfort, 2024-06-22:
I assume you were able to run your design. If not we can reopen this issue.

closed 2024-06-22 by gadfort