siliconcompiler/siliconcompiler #2059: siliconcompiler._common.SiliconCompilerError: These final steps could not be reached: ['export']

vjkr, 2023-10-26:
Hi,
I am trying to run example demo heartbeat. I observed the following results.
```
sc -target asic_demo -remote

and 

import siliconcompiler
chip = siliconcompiler.Chip('heartbeat')
chip.load_target('freepdk45_demo')
chip.input('/content/siliconcompiler/examples/heartbeat/heartbeat.v')
chip.clock('clk', period=10)
chip.set('option','remote', True)
chip.run()
````
Above code succeeds

But
```
!python3 /content/siliconcompiler/examples/heartbeat/heartbeat.py 
```
Fail with below logs
```
| INFO    | /content/siliconcompiler/examples/heartbeat/heartbeat.v inferred as rtl/verilog
| INFO    | /content/siliconcompiler/examples/heartbeat/heartbeat.sdc inferred as constraint/sdc
| WARNING | Environment variable $DISPLAY or $WAYLAND_DISPLAY not set
| WARNING | Setting ['option', 'nodisplay'] to True
| INFO    | job0 | --        | - | Setting up node import0 with surelog/parse
| INFO    | job0 | --        | - | Setting up node syn0 with yosys/syn_asic
| INFO    | job0 | --        | - | Setting up node floorplan0 with openroad/floorplan
| INFO    | job0 | --        | - | Setting up node place0 with openroad/place
| INFO    | job0 | --        | - | Setting up node cts0 with openroad/cts
| INFO    | job0 | --        | - | Setting up node route0 with openroad/route
| INFO    | job0 | --        | - | Setting up node dfm0 with openroad/dfm
| INFO    | job0 | --        | - | Setting up node export0 with klayout/export
| INFO    | job0 | --        | - | Setting up node export1 with openroad/export
| INFO    | job0 | --        | - | Checking manifest before running.
| INFO    | job0 | import    | 0 | Tool 'surelog' found with version '1.75' in directory '/usr/local/lib/python3.10/dist-packages/siliconcompiler/tools/surelog/bin'
| INFO    | job0 | import    | 0 | Running in /content/build/heartbeat/job0/import/0
| INFO    | job0 | import    | 0 | surelog -nocache +libext+.sv+.v -parse -nouhdm /content/siliconcompiler/examples/heartbeat/heartbeat.v -top heartbeat
| INFO    | job0 | import    | 0 | [INF:CM0023] Creating log file "/content/build/heartbeat/job0/import/0/slpp_all/surelog.log".
| INFO    | job0 | import    | 0 | [WRN:PA0205] /content/siliconcompiler/examples/heartbeat/heartbeat.v:1:1: No timescale set for "heartbeat".
| INFO    | job0 | import    | 0 | [INF:CP0300] Compilation...
| INFO    | job0 | import    | 0 | [INF:CP0303] /content/siliconcompiler/examples/heartbeat/heartbeat.v:1:1: Compile module "work@heartbeat".
| INFO    | job0 | import    | 0 | [INF:CP0302] Compile class "work@mailbox".
| INFO    | job0 | import    | 0 | [INF:CP0302] Compile class "work@process".
| INFO    | job0 | import    | 0 | [INF:CP0302] Compile class "work@semaphore".
| INFO    | job0 | import    | 0 | [INF:EL0526] Design Elaboration...
| INFO    | job0 | import    | 0 | [NTE:EL0503] /content/siliconcompiler/examples/heartbeat/heartbeat.v:1:1: Top level module "work@heartbeat".
| INFO    | job0 | import    | 0 | [NTE:EL0508] Nb Top level modules: 1.
| INFO    | job0 | import    | 0 | [NTE:EL0509] Max instance depth: 1.
| INFO    | job0 | import    | 0 | [NTE:EL0510] Nb instances: 1.
| INFO    | job0 | import    | 0 | [NTE:EL0511] Nb leaf instances: 0.
| INFO    | job0 | import    | 0 | [INF:UH0706] Creating UHDM Model...
| INFO    | job0 | import    | 0 | [  FATAL] : 0
| INFO    | job0 | import    | 0 | [ SYNTAX] : 0
| INFO    | job0 | import    | 0 | [  ERROR] : 0
| INFO    | job0 | import    | 0 | [WARNING] : 1
| INFO    | job0 | import    | 0 | [   NOTE] : 5
| WARNING | job0 | import    | 0 |  2: [WRN:PA0205] /content/siliconcompiler/examples/heartbeat/heartbeat.v:1:1: No timescale set for "heartbeat".
| INFO    | job0 | import    | 0 | Finished task in 0.99s
| ERROR   | job0 | syn       | 0 | Cannot resolve path ../third_party/pdks/virtual/freepdk45/libs/nangate45/r1p0/lib/NangateOpenCellLibrary_typical.lib in required file keypath ['library', 'nangate45', 'output', 'typical', 'nldm'].
| ERROR   | job0 | syn       | 0 | Cannot resolve path ../third_party/pdks/virtual/freepdk45/libs/nangate45/r1p0/techmap/yosys/cells_adders.v in required file keypath ['library', 'nangate45', 'option', 'file', 'yosys_addermap'].
| ERROR   | job0 | syn       | 0 | Fatal error in check_manifest()! See previous errors.
| ERROR   | job0 | syn       | 0 | Halting step 'syn' index '0' due to errors.
Traceback (most recent call last):
  File "/content/siliconcompiler/examples/heartbeat/heartbeat.py", line 17, in <module>
    main()
  File "/content/siliconcompiler/examples/heartbeat/heartbeat.py", line 11, in main
    chip.run()                                # run compilation
  File "/usr/local/lib/python3.10/dist-packages/siliconcompiler/core.py", line 3999, in run
    self._local_process(flow, status)
  File "/usr/local/lib/python3.10/dist-packages/siliconcompiler/core.py", line 3933, in _local_process
    self._check_nodes_status(flow, status)
  File "/usr/local/lib/python3.10/dist-packages/siliconcompiler/core.py", line 3890, in _check_nodes_status
    self.error(f'These final steps could not be reached: {list(unreachable_steps)}',
  File "/usr/local/lib/python3.10/dist-packages/siliconcompiler/core.py", line 4803, in error
    raise SiliconCompilerError(msg) from None
siliconcompiler._common.SiliconCompilerError: These final steps could not be reached: ['export']
```

I can see that PDKs are not resolved.
Please help me in this matter. Thank you in advance

vjkr, 2023-10-27:
> You may need to set the SCPATH correctly: https://docs.siliconcompiler.com/en/latest/user_guide/installation.html#install-from-pypi-org
Hi, It seems I am unable to set the path. I tried the below combinations(where 'openroad' is my username). But end up with the same error. 
```
export SCPATH=/home/openroad/Desktop/
export SCPATH=/home/openroad/Desktop/siliconcompiler
```
I try to run
```
(venv) openroad@ubuntu:~/Desktop/siliconcompiler/siliconcompiler/examples/heartbeat$ python3 heartbeat.py
```
Below is my directory structure
```
├── openlane_build_script
│   └── work
│       └── tools
├── Physical-Design-with-OpenLANE-using-SKY130-PDk
│   ├── images
│   └── picorv32a
│       ├── configs
│       ├── Results
│       └── source
└── siliconcompiler
    ├── build
    │   └── heartbeat
    ├── siliconcompiler
    │   ├── build
    │   ├── docs
    │   ├── examples
    │   ├── setup
    │   ├── siliconcompiler
    │   ├── tests
    │   ├── third_party
    │   └── venv
    └── venv
        ├── bin
        ├── etc
        ├── include
        ├── lib
        ├── lib64 -> lib
        └── share
```
Can you please guide me? Thank You.

cadnav, 2023-10-27:
You need to install the required packages prior to running the program.
Also, I would recommend using ChatGPT for immediate resolution for error
messages.
It will be a slow and painful learning curve if your path to resolving
issues is posting issues on the developer’s repo. Thank me later

Nado15, 2023-11-16:
> Hi, It seems I am unable to set the path. I tried the below combinations(where 'openroad' is my username). But end up with the same error.
>
> ```
> export SCPATH=/home/openroad/Desktop/
> export [...]

I had the same problem as you but your whole path needs to be ("Siliconcompiler" 2 times in a row)

`export SCPATH=/home/openroad/Desktop/siliconcompiler/siliconcompiler/`

gadfort, 2023-11-17:
With the release of 0.17.0 the SCPATH has been deprecated and removed, which should avoid this problem all together. Right now the remote is not updated to handle requests from 0.17.0, but that should be happening soon

mahati-basavaraju, 2024-04-25:
> Hi, I am trying to run example demo heartbeat. I observed the following results.
>
> ```
> sc -target asic_demo -remote
>
> and
>
> import siliconcompiler
> chip = siliconcompiler.Chip('heartbeat') [...]

Hi @vjkr , I am facing the same issue. Were you able to resolve this?

gadfort, 2024-04-25:
@vjkr what version of siliconcompiler are you using? The nangate45 files it cannot find were moved a while ago.

mahati-basavaraju, 2024-04-25:
@gadfort I have installed the logik fpga toolchain via pip. Therefore, the siliconcompiler was installed as a prerequisite. Therefore, the version I have is 0.21.7

mahati-basavaraju, 2024-04-25:
I get a similar error when trying to run the adder example in logik, please refer to the image below. This error occurs in the last stages while running silicon compiler.
[image]

gadfort, 2024-04-25:
Try adding `print(siliconcompiler.__version__)` to the `!python3 /content/siliconcompiler/examples/heartbeat/heartbeat.py` something seems off about that.
I'll take a look at the logs to see what happened on the remote run.

mahati-basavaraju, 2024-04-25:
Tried printing the version inside heartbeat.py file. Still shows the same version in use.
[image]

gadfort, 2024-04-25:
Could you try:
`pip install --upgrade siliconcompiler`
I there is a version mismatch that is causing an issue with the remote run.

mahati-basavaraju, 2024-04-25:
I upgraded it to 0.21.10 but the error persists. Also, it says that this version isn't compatible with logik.
[image]

gadfort, 2024-04-25:
The version incompatibility is not an issue (it will be fixed in the next release of logik). There is however a bug in the example.
Lines 10 and 11 should be corrected:
From:
```
    chip.input(f'{root}/heartbeat.v')                 # define list of source files
    chip.input(f'{root}/heartbeat.sdc')               # set constraints file
```
To:
```
    chip.input(os.path.join(root, 'heartbeat.v'))                 # define list of source files
    chip.input(os.path.join(root, 'heartbeat.sdc'))               # set constraints file
```

I'll fix that

mahati-basavaraju, 2024-04-25:
Thanks much @gadfort !

gadfort, 2024-04-25:
The examples are fixed in: https://github.com/siliconcompiler/siliconcompiler/pull/2373

closed 2024-05-23 by gadfort